Wiz-IO/platform-quectel #1: issue creating a project

mampez, 2019-01-24:
Hi guys,

First, let me congratulate you, it's a great job the possibility of using BC66 via PlatformIO.

I got this error message when I tried to create a project. Could you please help?

PIO Core Call Error: "The next files/directories have been created in C:\\Users\\mencho\\Documents\\PlatformIO\\Projects\\Test_BC66\r\ninclude - Put project header files here\r\nlib - Put here project specific (private) libraries\r\nsrc - Put project source files here\r\nplatformio.ini - Project Configuration File\n\nError: Could not find a version that satisfies the requirement '>=4.8.3' for your system 'windows_x86'"

Best regards,

M

Wiz-IO, 2019-01-24:
Hi, Manuel
thanks...

what is the version 4.8.3

sometime PIO say similar errors on Create New... 
but project is created, try Add project folder to workspace
or try again Create New

mampez, 2019-01-25:
Solved. Reinstalled everything and runs perfect!