Wiz-IO/platform-quectel #3: Cannot upload to BC66

hunasdf, 2019-03-20:
Hi,

We have a BC66-TE-B board and we would like to programming it with PIO and OpenCPU. If we install the Quectel platform as the documentation writes, we got " arm-none-eabi-gcc: error: unrecognized command line option '-R' " error. I fixed it with:
1. Removed the installed Quectel platform
2. Deleted the "-Rbuild" linkflag from the opencpu_bc66.py file (line 59th)
3. Reinstall the modified Quectel platform

Then, the code (the example from the WIKI) builds successfully, and starts to upload, but nothing happens during the upload. Cannot see any progress and never finish. No error or anything. We tired with bc66 and bc66-te-b board also, but the problem still. Have you any idea what should we do? How can we upload the code?

And an other interesting thing: we cannot update the firmware or create a backup from it. The port drivers looks good. Furthermore, the "EEPROM Programmer v1.7a" program also cannot read or write any value from/to the eeprom. But if I am logging out the serial port, and press the pwr_key, some information appear on the terminal.

We use OSX 10.14.3 and Visual Studio Code 1.32.3 with PIO Home 2.0.0 and Core 3.6.5 to build and upload, and Win7 x64 with VMWare Fusion to update or backup the firmware.

I attached few screenshots and logs. I hope this will

 help on us. But if you need any other info/log/data, I will try to supplement with it.

Thanks for any help!
Gergo

[upload_log.txt](https://github.com/Wiz-IO/platform-quectel/files/2987433/upload_log.txt)
[upload_log_verbose.txt](https://github.com/Wiz-IO/platform-quectel/files/2987434/upload_log_verbose.txt)
[image]
[image]
[image]

Wiz-IO, 2019-03-20:
Hi
for option -R I know

board switcher (uart selection) must be inward for USB

For macOS maybe need usb drivers

for manual COM port selection - project INI
[env:bc66]
platform = quectel
board = bc66
framework = arduino
;board_build.firmware = BC66R01A04V01
**upload_port = COM23
monitor_port = COM23**
monitor_speed = 115200

hunasdf, 2019-03-20:
Thanks for the fast reaction!

Sorry, I forget: the switch on the board always was "MAIN UART TO USB".

I installed the linked driver. Nothing changed... The 2 serial ports detected as default.
[image]

Tried with manual port selection:
[env:bc66-te-b]
platform = quectel
board = bc66-te-b
framework = opencpu
build_flags = -D DEBUG=1
upload_port = /dev/cu.usbmodem14201 ;or /dev/cu.usbmodem14203
monitor_port = /dev/cu.usbmodem14201 ;or /dev/cu.usbmodem14203
monitor_speed = 115200

and with Arduino framework:
[env:bc66-te-b]
platform = quectel
board = bc66-te-b
framework = arduino
build_flags = -D DEBUG=1
upload_port = /dev/cu.usbmodem14201 ;or /dev/cu.usbmodem14203
monitor_port = /dev/cu.usbmodem14201 ;or /dev/cu.usbmodem14203
monitor_speed = 115200

but the problem still. The upload never finish :(

Wiz-IO, 2019-03-20:
Try AT commands via terminal
the original Mediatek/Quectel FlashTool.exe work?
Can you test on Windows?

Wiz-IO, 2019-03-20:
com port work, must work and FlashTool.exe

hunasdf, 2019-03-20:
Thank you very much! Working on Windows 7, but not on OSX. And the main problem was that I have to push the reset and/or power button to start the upload. Unfortunately, the "Wait for Power ON or Reset module ..." message appear after I push one button, not before. Therefore, I will take a try on Mac too, if I have a little time.
Thank you again!

Wiz-IO, 2019-03-20:
YW
I test macOS from virtual machine(win...driver)
I will change message

closed 2019-09-27 by Wiz-IO