Wiz-IO/platform-quectel #4: BC26 module cannot upload @ linux VScode

Senrecon, 2019-03-25:
Some upload openCPU BC26 flash error:

Short press 
Auto-detected: /dev/ttyUSB0
Uploading: program
Wait for Power ON or Reset module <.........................................> DONE
Starting <..........
ERROR: DA 1 status
*** [upload] Explicit exit, status 2
===================================== [ERROR] Took 5.91 seconds =====================================
The terminal process terminated with exit code: 1

Long Press
Auto-detected: /dev/ttyUSB0
Uploading: program
Wait for Power ON or Reset module <..................> DONE
Starting <....................................................................................................................
ERROR: baudrate CONF
*** [upload] Explicit exit, status 2
===================================== [ERROR] Took 1.64 seconds =====================================
The terminal process terminated with exit code: 1

I m not use BC66-TE-B, I used another board.

I'm trying to change the repeat time of     

def da_start(self, nvdm = False):   from 50 to 500, it still failed to upload. 

When I use windows MTK IOT Flashtool.exe, following the guidance, it succ to upload.

http://doc.iotxx.com/NB-QuickStarter%E4%BD%BF%E7%94%A8%E8%AF%B4%E6%98%8E%E4%B9%A6 

system diagram like below:
http://doc.iotxx.com/%E6%96%87%E4%BB%B6:%E5%8A%9F%E8%83%BD%E6%A1%86%E5%9B%BE-NBQS.png

Wiz-IO, 2019-03-25:
for Linux maybe need drivers
https://www.maxlinear.com/product/interface/uarts/usb-uarts/xr21v1412

If Flashtool.exe can comunicate you can load cfg/bin from project folder
....PROJECT\.pioenvs\bc66\program.cfg

For BC26 work only OpenCPU for now
For Arduino you need BC66

Senrecon, 2019-03-25:
Did you mean the linux auto-installed driver cannot work?

I worked at OpenCPU on BC26 now.

Wiz-IO, 2019-03-25:
I have questions from others that there is a problem with the driver under Linux
For now use Windows... later I will test all under linux

Senrecon, 2019-03-25:
Thanks ! 
I will try to update CH340 USB2Uart driver @ linux, to see if it can work?
will update this issue later.

Wiz-IO, 2019-03-25:
ok, test please
if you have BC26 firmware with DBG folder, please send me to make port for arduino

Senrecon, 2019-03-25:
Still cannot upload and blocked at Baudrate error.

I used following CH34x

https://github.com/juliagoda/CH341SER

Senrecon, 2019-03-25:
I only got OpenCPU firmware, no Dbug folder I think.

the download link:
https://pan.baidu.com/s/1OF3uIphvs3vrMHy2AHxREQ#list/path=%2Fsharelink1824053184-636263183088586%2FNB260%2FOpenCPU_SDK&parentPath=%2Fsharelink1824053184-636263183088586

Wiz-IO, 2019-03-25:
Are you sure you PowerON/Reset the module on boot

Senrecon, 2019-03-25:
On my board, PowerOn/Reset control by a jumper.I used PowerKey, on normal status, the powerkey will short to ground, when push the button, it will be 3.3v connected.

I'm not sure your point, at windows enviroment, it works normal.
at Linux, the debug info can be show at vscode terminal, but download always failed.

Wiz-IO, 2019-03-25:
try on Linux AT commands

Senrecon, 2019-03-25:
I will try tomorrow, thanks

Senrecon, 2019-03-26:
I shift to win vscode + platformio, it works normal.
I think this linux upload issue maybe not complicate, if follow MT2625 upload procedure(Document is better), root  should be dig out. need check feedback bytes when set baudrate 921600.
If you have document , would you please share with me, I will check it.

below is the win upload 
Checking size .pioenvs\bc26\program.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]   5.4% (used 5449 bytes from 100000 bytes)
PROGRAM: [          ]   0.6% (used 1120 bytes from 200000 bytes)
Configuring upload protocol...
Looking for upload port...
Auto-detected: COM7
Uploading: program
Wait for Power ON or Reset module <.......................> DONE
Starting <.........................................................> DONE
Uploading <.....................................> DONE

Wiz-IO, 2019-03-26:
https://github.com/Wiz-IO/platform-quectel/blob/master/builder/frameworks/pyMT2625.py#L285

I have not Mediatek documents... reverse engineering

Senrecon, 2019-03-26:
That's not easy, I just checked codes,and modified some print out to debug
I found 
1. the baudrate didnot rise up to 921600, just keep 115200
2. write CONF 0x69, get feedback 0x6D

I cannot tell what happened inside the module.
I have no idea to debug more.

Wiz-IO, 2019-03-26:
if work for you - skip rice the baudrate

Senrecon, 2019-03-26:
It cannot work.
I need convert to win dev environment.

DurandA, 2019-09-18:
> For BC26 work only OpenCPU for now
> For Arduino you need BC66

I confirm that you can flash BC26 with BC66 firmware!

closed 2019-09-27 by Wiz-IO