Wiz-IO/platform-quectel #5: About the math include

Senrecon, 2019-03-26:
when I include math.h and use cos() function, the compiling will pass.
but when run the app at opencpu(BC26), it always crashed and reboot.
I'm not sure if need add some gcc path of include into c_cpp_properties.json 

void proc_main_task(s32 taskId)
{
    s32 ret;
    u32 cnt = 0; 
    double a = 0;
    double b = 0;

    char Buff[10]="0.345";
    char* pStr;

    // Register & open UART port
    ret = Ql_UART_Register(m_myUartPort, CallBack_UART_Hdlr, NULL);
    if (ret < QL_RET_OK)
    {
        Ql_Debug_Trace("Fail to register serial port[%d], ret=%d\r\n", UART_PORT1, ret);
    }
    ret = Ql_UART_Open(m_myUartPort, 115200, FC_NONE);
    if (ret < QL_RET_OK)
    {
        Ql_Debug_Trace("Fail to open serial port[%d], ret=%d\r\n", UART_PORT1, ret);
    }
  
    
    a = cos(b); //crashed here

Wiz-IO, 2019-03-26:
huh...
for me math work BC66NBR01A04V01 
#include <math.h>
Serial.printf("cos(3.14) = %f\n", cos(3.14));
result:
cos(3.14) = -0.999999

I not have BC26 for tests
BC66 math is:
            "-mfloat-abi=hard",
            "-mfpu=fpv4-sp-d16",

Senrecon, 2019-03-26:
Do you have BC26 SDK?

Wiz-IO, 2019-03-26:
NO, I can not download SDK from baidu.com

Wiz-IO, 2019-03-26:
#include "custom_feature_def.h"
#include "ql_type.h"
#include "ql_stdlib.h"
#include "ql_uart.h"
#include "ril.h"
#include <math.h>

void CB_UART_Hdlr(Enum_SerialPort port, Enum_UARTEventType msg, bool level, void *customizedPara) {}
char DBG_BUFFER[512];
#define DBG(FORMAT, ...)                                                    \
    {                                                                       \
        Ql_memset(DBG_BUFFER, 0, 512);                                      \
        Ql_sprintf(DBG_BUFFER, FORMAT, ##__VA_ARGS__);                      \
        Ql_UART_Write(UART_PORT0, (u8 *)DBG_BUFFER, Ql_strlen(DBG_BUFFER)); \
    }

void proc_main_task(s32 taskId)
{
    s32 ret;
    Ql_UART_Register(UART_PORT0, CB_UART_Hdlr, NULL);
    Ql_UART_Open(UART_PORT0, 115200, FC_NONE);
    DBG("BEGIN\n");
    DBG("cos(3.14) = %f\n", cos(3.14));
    ST_MSG msg;
    while (1)
    {
        Ql_OS_GetMessage(&msg);
    }
}

result: 
F1: 0000 0000
V0: 0000 0000 [0001]
00: 0006 000C
01: 0000 0000
U0: 0000 0001 [0000]
T0: 0000 00B4
Leaving the BROM
BEGIN
cos(3.14) = -0.999999

Wiz-IO, 2019-03-26:
BTW: you can test to flash your BC26 with BC66 firmware ( but first make NVDM backup )
BC66NBR01A04V01 or BC66NBR01A05V01
I do not think there are any differences...

Senrecon, 2019-03-26:
I had upload SDK to:
https://drive.google.com/file/d/1MWJjCzdnAu1rHSiF1FXQ83ovcvXXQMoT/view

Wiz-IO, 2019-03-26:
I get it but I cannot see differences in make file
I use standart GCC for M4 and work fine with M66, MC60, BC66, BG96

Try compile with Quectel SDK for cos() test

Senrecon, 2019-03-26:
Thanks!
Would please share the BC66 SDK with me?

I had checked the harware datasheet, seems BC66 is same as BC66, only some band support is diff.

Wiz-IO, 2019-03-26:
email me: the.wizarda[AT]gmail.com

Senrecon, 2019-03-27:
Had sent your mail, 
some question: for BC26 OPENCPU : does the fota option included but no related folder and source in the  git?             join(CORE_DIR, "fota", "inc"),

Wiz-IO, 2019-03-27:
https://cloud.comet.bg/index.php/s/mc49pbHtL7fBP5P?path=%2FBC66%20Documents
BC66 SDK has no folder fota and Quectel DFOTA packer is not ready

Senrecon, 2019-03-27:
got it , thanks!

I had checked SDK, there is a line in make file  def 
ENV_INC='$(strip $(GCC_INSTALL_PATH))/arm-none-eabi/include'
I dont know how to check my ENV 's include path of gcc, can you show me ?

another strange thing is :
Even I comment #include <math.h> like:
//#include <math.h>
the compiling can go down without any error or warning.

Wiz-IO, 2019-03-27:
Are you have other GCC platforms in PlatformIO IDE
Maybe IDE link to other compiler without libgcc for hard float 

I ask PlatformIO how to set manual gcc path

there is math and gcc libs search path
https://github.com/Wiz-IO/platform-quectel/blob/master/builder/frameworks/opencpu_bc66.py#L68

Senrecon, 2019-03-27:
Thanks!
I used ESP32 with platformIO before, I m not sure if some GCC env is mixed-up .
Another thing is I had used SDK to build APP, seems it works fine:

+CFUN: 1
OpenCPU: float test !
cos(3.14) = -0.999999

Wiz-IO, 2019-03-27:
nice :) how?

is "mixed"https://github.com/Wiz-IO/platform-quectel/blob/master/platform.json#L43

Senrecon, 2019-03-27:
The platform.json is no problem from git downloaded, and I had follow your instruction paste the git link to advanced Platform to get it.

but for the LIBS=[ "gcc", "m", "app_start" ], 
https://github.com/Wiz-IO/platform-quectel/blob/master/builder/frameworks/opencpu_bc66.py#L68
I cannot tell if it include the header files of include dir

Senrecon, 2019-03-27:
Nothing Changed, still failed.
1. I had checked the win system path , no error or ESP32 path included.
2. I had check the VScode workspace's "Path" ," Include“， Nothing related with ESP32‘s Xtensa gcc.
3. I had discard this workspace, generate a new project, builded, still failed.

there is the build log:(example_float_math.c include the main entry, used math.h, but building show No dependencies ,strange!!)
struggling......   any suggestion???

Verbose mode can be enabled via `-v, --verbose` option
<<<<<<<<<<<< QUECTEL BC26 MODULE 2018 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/quectel/bc26.html
PLATFORM: Quectel > Quectel BC26 module
HARDWARE: ARM, MT2625 78MHz 97.66KB RAM (195.31KB Flash)
DEBUG: CURRENT(uart) ON-BOARD(uart)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 0 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs\bc26\src\example_float_math.o
Compiling .pioenvs\bc26\framework\custom_sys_cfg.o
Compiling .pioenvs\bc26\framework\ril\src\ril_atResponse.o
Compiling .pioenvs\bc26\framework\ril\src\ril_init.o
Compiling .pioenvs\bc26\framework\ril\src\ril_lwm2m.o
Compiling .pioenvs\bc26\framework\ril\src\ril_network.o
Compiling .pioenvs\bc26\framework\ril\src\ril_onenet.o
Compiling .pioenvs\bc26\framework\ril\src\ril_socket.o
Compiling .pioenvs\bc26\framework\ril\src\ril_system.o
Compiling .pioenvs\bc26\framework\ril\src\ril_urc.o
Compiling .pioenvs\bc26\framework\ril\src\ril_util.o
Compiling .pioenvs\bc26\framework\sys_config.o
Compiling .pioenvs\bc26\framework\ril\ril_atResponse.o
Compiling .pioenvs\bc26\framework\ril\ril_init.o
Compiling .pioenvs\bc26\framework\ril\ril_lwm2m.o
Archiving .pioenvs\bc26\libframework.a
Compiling .pioenvs\bc26\framework\ril\ril_network.o
Compiling .pioenvs\bc26\framework\ril\ril_onenet.o
Compiling .pioenvs\bc26\framework\ril\ril_socket.o
Compiling .pioenvs\bc26\framework\ril\ril_system.o
Compiling .pioenvs\bc26\framework\ril\ril_urc.o
Compiling .pioenvs\bc26\framework\ril\ril_util.o
Archiving .pioenvs\bc26\framework\libril.a
Linking .pioenvs\bc26\program.elf
Checking size .pioenvs\bc26\program.elf
Building .pioenvs\bc26\program.dat
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [=         ]   5.4% (used 5445 bytes from 100000 bytes)
PROGRAM: [          ]   0.6% (used 1116 bytes from 200000 bytes)

Wiz-IO, 2019-03-27:
open your file
https://github.com/Wiz-IO/platform-quectel/blob/master/platform.json#L43
and change to
"version": "=1.70201.0"
delete folder toolchain-gccarmnoneeabi
build
platformIO must download version 1.70201.0

Senrecon, 2019-03-27:
Still crashed!
Changed according to your suggestion.
Then the gccarmnoneeabi download the new "version": "=1.70201.0"
the building result and size is same as before, no change, then I clean and rebuild it, Crash without any change.
I used the verbose build, the same crash and same size of DATA and Program size.

Wiz-IO, 2019-03-27:
send me the APP .bin .cfg and .elf file

Senrecon, 2019-03-27:
[bc26.zip](https://github.com/Wiz-IO/platform-quectel/files/3013489/bc26.zip)
Attached the ZIP file include  all files

Wiz-IO, 2019-03-27:
Leaving the BROM
OpenCPU: float test !
and crash :)
give me c file

Senrecon, 2019-03-27:
[SDK_build_BC26.zip](https://github.com/Wiz-IO/platform-quectel/files/3013519/SDK_build_BC26.zip)
The c file also included in the zip file, and all SDK build result(which can run on BC26 without crash)

Wiz-IO, 2019-03-27:
:)
APP BEGIN
MATH TEST
   cos(3) = -0.989992
  acos(3) = nan
   sin(3) = 0.141120
  asin(3) = nan
sum=116.691290
atof1=116.123 atof2=0.567800 var=58.000000 sum=10.567800
atof1=116.123 atof2=0.567800 var=58.000000
Float test, a=0.85,b=0.15,radLat1=31.110,radLat2=121.290, s=9.00000
Float test, pos=0.845889
Float test, a=0.85,b=0.15,radLat1=31.110,radLat2=121.290, s=9.00000
dst = 9.282km
0 MSG: 4097 0 0
Ril Ready

Wiz-IO, 2019-03-27:
[main.zip](https://github.com/Wiz-IO/platform-quectel/files/3013567/main.zip)

test pls

Senrecon, 2019-03-27:
Used main.c to compile-build: still crash.
 
genie connected: Yes

 auto-reboot: Yes

F1: 0000 0000
V0: 0000 0000 [0001]
00: 0006 000C
01: 0000 0000
U0: 0000 0001 [0000]
T0: 0000 00B4
Leaving the BROM

RDY

+CFUN: 1
APP BEGIN
MATH TEST

 ------------- Exception called -------------
 genie connected: Yes

 auto-reboot: Yes

Wiz-IO, 2019-03-27:
upload my program.bin from zip

result from your APPGS3MDM32A01
OpenCPU: float test !
cos(3.14) = -0.999999
sum=116.691290
atof1=116.123 atof2=0.567800 var=58.000000 sum=10.567800
atof1=116.123 atof2=0.567800 var=58.000000
Float test, a=0.85,b=0.15,radLat1=31.110,radLat2=121.290, s=9.00000
Float test, pos=0.845889
Float test, a=0.85,b=0.15,radLat1=31.110,radLat2=121.290, s=9.00000
dst = 9.282km

Wiz-IO, 2019-03-27:
this is my settings

    env.Append(
       CPPDEFINES = [ # -D                         
            "CORE_" + core.upper().replace("-", "_"),
            platform.upper(),
        ],        
        CPPPATH = [ # -I
            join(framework_dir, "opencpu", core),
            join(framework_dir, "opencpu", core, "include"),
            join(framework_dir, "opencpu", core, "ril", "inc"),   
            join("$PROJECT_DIR", "lib"),
            join("$PROJECT_DIR", "include"),
            join("$PROJECT_DIR", "config")      
        ],         
        CFLAGS = [
            "-Os", "-g", "-c",        
            "-mcpu=cortex-m4",
            "-mfloat-abi=hard",
            "-mfpu=fpv4-sp-d16", 
            "-mlittle-endian",      
            "-mthumb",  
            "-std=c11", 
            #"-fno-builtin",                 
            "-fdata-sections",      
            "-ffunction-sections",
            "-fno-strict-aliasing",
            "-fsingle-precision-constant",             
            "-Wall", 
            "-Wextra",           
            "-Wno-unused",
            "-Wno-unused-parameter",              
            "-Wno-sign-compare",  
            "-Wno-pointer-sign",
            #"-Wno-pointer-compare",
            #"-Wno-type-limits",
            #"-Wno-int-conversion",
            #"-Wno-missing-braces",
            #"-Wno-missing-field-initializers",
                    
        ],        
        LINKFLAGS = [    
            "-mcpu=cortex-m4",
            "-mfloat-abi=hard",
            "-mfpu=fpv4-sp-d16", 
            "-mlittle-endian",      
            "-mthumb",                                             
            "-nostartfiles",   
            "-fno-use-cxa-atexit",         
            "-Xlinker", "--gc-sections",              
            "-Wl,--gc-sections",
        ],    
        LIBPATH = [ lib_dir ],      
        LDSCRIPT_PATH = linker, 
        LIBS = [ "gcc", "m", "_app_start_{}".format(core), ],

Senrecon, 2019-03-27:
I just noticed your setting is a little big diff with opencpu_bc66.py, which like below(I used):

def bc66_init(env):
    DIR = os.path.dirname(env.get("BUILD_SCRIPT"))
    TOOL_DIR = env.PioPlatform().get_package_dir("tool-quectel")
    CORE = env.BoardConfig().get("build.core") # "bc66"
    CORE_DIR = join(env.PioPlatform().get_package_dir("framework-quectel"), "opencpu", CORE)
    env.Append(
        CPPDEFINES=["OPENCPU", "CORE_"+CORE.upper(),  "_REENT_SMALL"], # -D
        CPPPATH=[ # -I
            CORE_DIR,
            join(CORE_DIR, "include"),
            join(CORE_DIR, "ril", "inc"),
            join(CORE_DIR, "fota", "inc"),        
        ],
        CFLAGS=[
            "-mcpu=cortex-m4",
            "-mfloat-abi=hard",
            "-mfpu=fpv4-sp-d16",
            "-fsingle-precision-constant",        
            "-mthumb",
            "-mthumb-interwork", 
            "-mlong-calls",       
            "-std=c99",
            "-c", "-g", "-Os",
            "-fno-builtin",         
            "-ffunction-sections",
            "-fdata-sections",
            "-fno-strict-aliasing",
            "-fno-common", 
            "-Wall",
            "-Wp,-w",               
            "-Wstrict-prototypes",                  
            "-Wno-implicit-function-declaration",
        ],

        LINKFLAGS=[        
            "-mcpu=cortex-m4",
            "-mfloat-abi=hard",
            "-mfpu=fpv4-sp-d16",
            "-mthumb",
            "-mthumb-interwork",   
            "-nostartfiles",     
            #"-Rbuild",        
            "-Wl,--gc-sections,--relax",
            "-Wl,-wrap=malloc",
            "-Wl,-wrap=calloc",
            "-Wl,-wrap=realloc",
            "-Wl,-wrap=free"   
        ],   
        LIBPATH=[CORE_DIR],
        LDSCRIPT_PATH=join(CORE_DIR, "linkscript.ld"), 
        LIBS=[ "gcc", "m", "app_start" ], 

        BUILDERS = dict(
            ElfToBin = Builder(
                action=env.VerboseAction(" ".join([
                    "$OBJCOPY",
                    "-O",
                    "binary",
                    "$SOURCES",
                    "$TARGET"
                ]), "Building $TARGET"),
                suffix=".dat"
            ),    
            MakeHeader = Builder( # Add BIN Header
                action = env.VerboseAction(bc66_header, "Adding GFH header"),
                suffix = ".bin"
            )       
        ), # dict
        
        UPLOADCMD = bc66_uploader         
    ) # env.Append    

    libs = []
    libs.append(
        env.BuildLibrary(
            join("$BUILD_DIR", "framework"),
            join( CORE_DIR ),
    ))
    libs.append(
        env.BuildLibrary(
            join("$BUILD_DIR", "framework", "ril"),
            join(CORE_DIR, "ril", "src")
    ))
    env.Append( LIBS = libs )

Wiz-IO, 2019-03-27:
yep, I know - this is my last port settings for new version
I update git soon
include full Arduino compatible ports for BC66, M66 and MC60

Senrecon, 2019-03-27:
Sure, wait for u update, it should be works, I just replace build argc items it in my be66.py.

It can works a little more:(I think this is root cause)

RDY

+CFUN: 1
APP BEGIN
MATH TEST
   _cos(3) = -0.989992_

 ------------- Exception called -------------
 genie connected: Yes

 auto-reboot: Yes

Senrecon, 2019-03-27:
BTW:still some link option failed to compile.
 would you please share the whole python file with me? 
I will check at my side.

Wiz-IO, 2019-03-27:
I found the problem:
.....platformio\packages\framework-quectel\opencpu\bc66
rem this ___libgcc.a

and replace this

        CPPDEFINES=[ "CORE_BC66", "OPENCPU"], # -D
        CPPPATH=[ # -I
            CORE_DIR,
            join(CORE_DIR, "include"),
            join(CORE_DIR, "ril", "inc"),
            join(CORE_DIR, "fota", "inc"),        
        ],
        CFLAGS=[
            "-Os", "-g", "-c",        
            "-mcpu=cortex-m4",
            "-mfloat-abi=hard",
            "-mfpu=fpv4-sp-d16", 
            "-mlittle-endian",      
            "-mthumb", 
            "-std=c11",                 
            "-fdata-sections",      
            "-ffunction-sections",
            "-fno-strict-aliasing",
            "-fsingle-precision-constant",             
            "-Wall", 
            "-Wextra",           
            "-Wno-unused",
            "-Wno-unused-parameter",              
            "-Wno-sign-compare",  
            "-Wno-pointer-sign",
        ],
        LINKFLAGS=[        
            "-mcpu=cortex-m4",
            "-mfloat-abi=hard",
            "-mfpu=fpv4-sp-d16", 
            "-mlittle-endian",      
            "-mthumb",                                             
            "-nostartfiles",        
            "-Xlinker", "--gc-sections",              
            "-Wl,--gc-sections",  
        ],   
        LIBPATH=[CORE_DIR],
        LDSCRIPT_PATH=join(CORE_DIR, "linkscript.ld"), 
        LIBS=[ "gcc", "m", "app_start" ],

Wiz-IO, 2019-03-27:
Math test work and with Arduino and OpenCPU
[arduino_math_test.zip](https://github.com/Wiz-IO/platform-quectel/files/3014351/arduino_math_test.zip)

Senrecon, 2019-03-28:
I dont understand waht's mean of rem ___libgcc.a 
I had tried rename it, build , seems succ.

Would you please give some explaination of libgcc.a ? why it cause such a issue?

RDY

+CFUN: 1
OpenCPU: float test !

sum=116.691290
atof1=116.123 atof2=0.567800 var=58.000000 sum=10.567800
atof1=116.123 atof2=0.567800 var=58.000000
Float test, a=0.85,b=0.15,radLat1=31.110,radLat2=121.290, s=9.00000
Float test, pos=0.845889
Float test, a=0.85,b=0.15,radLat1=31.110,radLat2=121.290, s=9.00000
dst = 9.282km

Wiz-IO, 2019-03-28:
remark, rename or delete...
This lib is from original SDK, but apparently does not work with this compiler

closed 2019-04-10 by Senrecon