Wiz-IO/platform-quectel #6: Compilation and Upload error + crosscompile library query - OpenCPU M66 @ linux vscode.

VijaiDanni, 2019-04-17:
Hi,
    First of all I haven't found any forum/community to post the query so i am sorry if i am doing this in the wrong place.
    I am working on an IOT related project using Quectel M66 module. Previously we were working in a windows environment using the official openCPU Quectel SDK for the M66 module. No problem there, worked real fine and was uploaded to the module without any errors. I have only seen your library yesterday and i have tried to port the project over to a linux environment. I haven't used platformIO before. I have installed the platformio visual studio IDE in linux and successfully installed all libraries based on your documentation. Copied the code over and hit build button in the platformIO Project Tasks and a few errors were thrown by the compiler. I have identified some of the issues and now passing it over to you so that you can resolve it from your end.

1) Include file name typo error in supplied RIL library.
The typos are same with the libraries supplied in windows Quectel M66 sdk. But SDK compiler didn't threw an error. Might be helpful for others attempting to port their code.
ril_dir - ~/.platformio/packages/framework-quectel/opencpu/m66/
a) #include "Lib_ril_sms.h"     - Line 39 in "ril_sms.h" should be               - #include "lib_ril_sms.h"
b) #include "ril.h "                   - Line 35 in "ril_atResponse.c" should be  - #include "ril.h"
c) #include "ril_telephony.h "  - Line 36 in "ril_telephony.c" should be       - #include "ril_telephony.h"
d) #include "ril_telephony.h "  - Line 38 in "ril_urc.c" should be                 - #include "ril_telephony.h"
c) #include "ril_util.h "             - Line 33 in "ril_util.c" should be                 - #include "ril_util.h"

2) Custom Feature definition.
Inorder to use sms, call, fota features, I need to enable the features in the custom_feature_def.h located in the project folder. If i use the file created during project initialization(with none of the features enabled), the error is thrown as follows.

.pioenvs/m66/src/main.o: In function `smsSend':
main.c:(.text+0x192): undefined reference to `RIL_SMS_SendSMS_Text_Ext'
.pioenvs/m66/src/main.o: In function `urc_incoming_call_handler':
main.c:(.text+0xc8e): undefined reference to `RIL_Telephony_Hangup'
.pioenvs/m66/src/main.o: In function `ParseSerialCommand':
main.c:(.text+0xe84): undefined reference to `RIL_Telephony_Dial'
.pioenvs/m66/src/main.o: In function `urc_incoming_sms_handler':
main.c:(.text+0x17a2): undefined reference to `RIL_SMS_ReadSMS_Text'
.pioenvs/m66/src/main.o: In function `Fota_Upgrade':
main.c:(.text.unlikely+0x4a): undefined reference to `Ql_FOTA_StartUpgrade'
collect2: error: ld returned 1 exit status
*** [.pioenvs/m66/program.elf] Error 1

The error doesn't even mention the existence of a config_feature_def.h. Or that I should check if necessary flags are enabled. Perhaps that can be resolved in the documentation. 
After enabling the flags the program compiled successfully.

3) upload to M66 device.
I have used the upload option in Project Tasks. The compiler gives me this error. Now how/where do i specify these options?

Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1

in windows sdk I could select all this in the gui program they supplied.
Parameters used in win - - baudrate - 115200. device name-M66FA. COM1

4) As for the query, Is it posssible to compile wolfssl/openssl for openCPU and use it along with platform-quectel? I need to create JWT token for google cloud iot authentication. Every available source mentions using external hardware ICs for obtaining the encrypted string.I am just trying to see if I could create a JWT token from M66 module itself. The jwt string needs to be signed by a key and thats where ssl library comes to play. 
    The truth is that I am more of a python programmer and I never tried crosscompiling before and don't have much time to spend, if it isn't productive. If it is possible can you mention how I can do that.

Thank you. And thank you again for this amazing work.

Wiz-IO, 2019-04-17:
Hi, thanks for this

1. This is Quectel bug - Linux case sensitive include path names... I use original openCPU and...I will re-edit GIT very soon

2. Look your PIO Project - folder "config" and edit features, I create this folder/sources at the first build as "template" for the project, if you delete this folder, builder will make new clear

3. I have bug with python uploader for M66/MC60... upload only small applications. I will investigate later

Open project platformio.ini and add lines
upload_port = COM21 ; <--- linux port name
monitor_port = COM21
monitor_speed = 115200

https://www.youtube.com/watch?v=JIvbUvgRH6k&t=29s

4. For now I not have solution for SSL ... later I will test to compile mbedtls

VijaiDanni, 2019-04-18:
Hi,
    Thanks for your prompt response. Your instructions worked. I was able to compile code and upload for Quectel M66.
    I have declared multiple tasks in my code. And it didn't work at first. But going through the other config files, I see that i had to declare the task function in file- custom_task_cfg.h present in the config folder.
    Still working on porting the entire code. You have mentioned "upload small application" - The bin file that i tested linux compiler functionality with was 24kb. My project until now has reached 80kb. It will increase as i add more functionality. Will that be a problem when uploading from platformio?
    I will update if cross-compiling wolfssl for Quectel M66 is successfull. Their documentation gives me hope. I will look at mbedtls too, maybe its easier.

Thanks.

Wiz-IO, 2019-04-18:
for now you need to use Windows Uploader :(

Wiz-IO, 2019-04-18:
Quectel bug - Linux case sensitive include path names...
**IS DONE** for BC66, M66, MC60

Wiz-IO, 2019-04-19:
**M66 mbedtls chiper test**

`
--- SSL TEST BEGIN ---
  MD5 test #1: passed
  MD5 test #2: passed
  MD5 test #3: passed
  MD5 test #4: passed
  MD5 test #5: passed
  MD5 test #6: passed
  MD5 test #7: passed

  SHA-1 test #1: passed
  SHA-1 test #2: passed
  SHA-1 test #3: passed

  SHA-224 test #1: passed
  SHA-224 test #2: passed
  SHA-224 test #3: passed
  SHA-256 test #1: passed
  SHA-256 test #2: passed
  SHA-256 test #3: passed

  ARC4 test #1: passed
  ARC4 test #2: passed
  ARC4 test #3: passed

  DES -ECB- 56 (dec): passed
  DES -ECB- 56 (enc): passed
  DES3-ECB-112 (dec): passed
  DES3-ECB-112 (enc): passed
  DES3-ECB-168 (dec): passed
  DES3-ECB-168 (enc): passed

  DES -CBC- 56 (dec): passed
  DES -CBC- 56 (enc): passed
  DES3-CBC-112 (dec): passed
  DES3-CBC-112 (enc): passed
  DES3-CBC-168 (dec): passed
  DES3-CBC-168 (enc): passed

  AES-ECB-128 (dec): passed
  AES-ECB-128 (enc): passed
  AES-ECB-192 (dec): passed
  AES-ECB-192 (enc): passed
  AES-ECB-256 (dec): passed
  AES-ECB-256 (enc): passed

  AES-CBC-128 (dec): passed
  AES-CBC-128 (enc): passed
  AES-CBC-192 (dec): passed
  AES-CBC-192 (enc): passed
  AES-CBC-256 (dec): passed
  AES-CBC-256 (enc): passed

  Base64 encoding test: passed
  Base64 decoding test: passed

  MPI test #1 (mul_mpi): passed
  MPI test #2 (div_mpi): passed
  MPI test #3 (exp_mod): passed
  MPI test #4 (inv_mod): passed
  MPI test #5 (simple gcd): passed

  RSA key validation: passed
  PKCS#1 encryption : failed
  X.509 certificate load: passed
  X.509 signature verify: failed
  CTR_DRBG (PR = TRUE) : failed
  ENTROPY test: failed
--- SSL TEST END --- TESTED: 12  FAILED: 4 
`
**ELAPSED: 180 seconds**

VijaiDanni, 2019-04-20:
Hi,
    Thanks for testing mbedtls. I am sure that it will suit our needs. Will update you as soon as i could.
    Will use windows uploader for now. Thanks for fixing the case-sensitive include path names. Your support is amazing :) .
Thanks.

Wiz-IO, 2019-04-20:
you welcome :)

closed 2019-09-27 by Wiz-IO