Wiz-IO/platform-quectel #9: Level converter

cskocsis, 2019-08-30:
Hello,
What level shifter circuit did you use for the I2C interface? Level translator IC or single N-channel MOSFET?

TravisJoe, 2019-08-30:
On I2C you should not need a level shifter unless the logic levels are very different. You would just use the pull up resistors to the lower of the voltages. This does assume that you do not have pull-up resistors set on the I2C pins.

Wiz-IO, 2019-08-30:
for tests I not use ... but is need... Mediatek chipset is "bad", Qualcomm is "good"
MOSFET or chip ... your choise

cskocsis, 2019-08-30:
If I understand well, OLED (3.3v) can be directly connected to BC66 (1.8V). Did you just use pull-up resistance?
https://raw.githubusercontent.com/Wiz-IO/platformio-quectel-examples/master/Arduino/bc66/i2c_oled/20190403_144444.jpg

Wiz-IO, 2019-08-30:
yep, but you can see the "noise" at the upper left corner

Wiz-IO, 2019-08-30:
Qualcomm chipset
https://media.licdn.com/dms/image/C5622AQH2qMvxKAJJkg/feedshare-shrink_8192/0?e=1570060800&v=beta&t=VfNIJP1Ea9oqOhbV2PI_1VycAR_PI4c0w0tygAiynlQ

closed 2019-08-30 by cskocsis

Wiz-IO, 2020-06-27:
OLED SSD1306 ( 3.3v ) work without level convertors
The "sparks" is bug in I2C and Adafruit library WIRE_MAX - M66,MC60, BC66 max transaction size is 8 bytes 

Adafruit_SSD1306.c
```
#define WIRE_MAX 8 
#ifndef  WIRE_MAX
/*
#if defined(BUFFER_LENGTH)
 #define WIRE_MAX BUFFER_LENGTH          ///< AVR or similar Wire lib
#elif defined(SERIAL_BUFFER_SIZE)
 #define WIRE_MAX (SERIAL_BUFFER_SIZE-1) ///< Newer Wire uses RingBuffer
#else
 #define WIRE_MAX 32                     ///< Use common Arduino core default
#endif
*/
#endif
```