Wiz-IO/platform-quectel #10: BC26 has BC66 core

DurandA, 2019-09-07:
On the BC26 platform, the referenced core is `bc66`:
https://github.com/Wiz-IO/platform-quectel/blob/207b87a64285f66ea6ccdc6f9e050210e8255044/boards/bc26.json#L5

I came here while investigating available NB-IoT platforms and I am confused. What are the differences between BC26 and BC66? From the specs they look the same so is one version a minor upgrade over the other? Also the BC26 seems to be more available on Chinese markets.

Wiz-IO, 2019-09-07:
BC26 is only for China, BC66 is international version
If you have BC26 module, try flash it with firmware for BC66 an test
I not have module for tests....

DurandA, 2019-09-08:
Thanks for the clarification.

I found a cheap [BC26 module on Electrodragon](https://www.electrodragon.com/product/nb-iot-network-module-bc26/). I will order one and give it a try. Will report my results there.

DurandA, 2019-09-14:
> BC26 is only for China, BC66 is international version
> If you have BC26 module, try flash it with firmware for BC66 an test
> I not have module for tests....

@Wiz-IO I just reveived the BC26 module. Is the [BC66NBR01A04V01](https://github.com/Wiz-IO/Arduino-Quectel-BC66/blob/master/firmwares/Firmware_BC66NBR01A04V01.zip) the latest B66 firmware? Is there a dedicated tool to flash the firmware?

Wiz-IO, 2019-09-14:
Hi, 
you need to connect to your local Quectel support for this

closed 2019-09-14 by DurandA

DurandA, 2019-09-18:
> BC26 is only for China, BC66 is international version
> If you have BC26 module, try flash it with firmware for BC66 an test
> I not have module for tests....

Hi @Wiz-IO!

Flashing the BC26 with BC66 firmware works fine, thanks! I flashed the version 4 firmware with success and ran the `info` Arduino example with success. Quectel sent me version 7 of the firmware with _QFlash_. _QFlash_ works fine just for flashing but doesn't have any of the advanced features of _IoT FlashTool_.

Does the official BC66 dev board have an auto reset? I have to very quickly power cycle the board in order to flash it. I think I destroyed the _rst_ pin by connecting it to the 5V as I didn't initially had the schematic of my board.

Finally, what tool is used to flash the firmware with PlatformIO? I had a look at platform.json but [tool-quectel](https://github.com/Wiz-IO/tool-quectel) repo is mostly empty. I expected some kind of bootstrapping with AT commands.
https://github.com/Wiz-IO/platform-quectel/blob/207b87a64285f66ea6ccdc6f9e050210e8255044/platform.json#L49-L53

Thank you very much for your open-source project(s).

Wiz-IO, 2019-09-19:
Hi :)
in time I remove exe tools and replace with python scripts (reverse)
here is BC flasher https://github.com/Wiz-IO/platform-quectel/blob/master/builder/frameworks/MT2625.py

The platform work on unix-like too

Wiz-IO, 2019-09-19:
BTW: Arduino port work only with firmwares 4 and 5 for now and not need other firmwares
I work over firmware 7 but I need time...
About OpenCPU - there are no limits, but  firmware 7 use SDK 1.4 - you can replace manualy
or wait my update