Wiz-IO/platform-quectel #13: Cannot create new project.

LaBus3, 2019-10-07:
`PIO Core Call Error: "The current working directory F:\\Workspace\\tesatstse will be used for the project.\n\r\nThe next files/directories have been created in F:\\Workspace\\tesatstse\ninclude - Put project header files here\nlib - Put here project specific (private) libraries\nsrc - Put project source files here\nplatformio.ini - Project Configuration File\n\n\nError: Processing ec2x (platform: quectel; board: ec2x; framework: arduino)\n-------------------------------------------------------------------------------\nVerbose mode can be enabled via `-v, --verbose` option\n<<<<<<<<<<<< QUECTEL EC2X MODULE 2018 Georgi Angelov >>>>>>>>>>>>\nCONFIGURATION: https://docs.platformio.org/page/boards/quectel/ec2x.html\nPLATFORM: Quectel 2.0.3 #fbd7a6e > Quectel EC2x module\nHARDWARE: ARM, MDM9607 1200MHz, 10.00MB RAM, 10.00MB Flash\nDEBUG: Current (uart) On-board (uart)\nPACKAGES: framework-quectel 44bdff7, toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)\nImportError: No module named arduino-ec2x:\n File \"C:\\users\\madhav\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\main.py\", line 126:\n env.SConscript(\"$BUILD_SCRIPT\")\n File \"C:\\Users\\Madhav\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 541:\n return _SConscript(self.fs, *files, **subst_kw)\n File \"C:\\Users\\Madhav\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 250:\n exec _file_ in call_stack[-1].globals\n File \"C:\\users\\madhav\\.platformio\\platforms\\quectel\\builder\\main.py\", line 21:\n elf = env.BuildProgram()\n File \"C:\\Users\\Madhav\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 224:\n return self.method(*nargs, **kwargs)\n File \"C:\\users\\madhav\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 110:\n env.BuildFrameworks(env.get(\"PIOFRAMEWORK\"))\n File \"C:\\Users\\Madhav\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Environment.py\", line 224:\n return self.method(*nargs, **kwargs)\n File \"C:\\users\\madhav\\.platformio\\penv\\lib\\site-packages\\platformio\\builder\\tools\\platformio.py\", line 291:\n SConscript(env.GetFrameworkScript(f), exports=\"env\")\n File \"C:\\Users\\Madhav\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 604:\n return method(*args, **kw)\n File \"C:\\Users\\Madhav\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 541:\n return _SConscript(self.fs, *files, **subst_kw)\n File \"C:\\Users\\Madhav\\.platformio\\packages\\tool-scons\\script\\..\\engine\\SCons\\Script\\SConscript.py\", line 250:\n exec _file_ in call_stack[-1].globals\n File \"C:\\users\\madhav\\.platformio\\platforms\\quectel\\builder\\frameworks\\arduino.py\", line 11:\n m = __import__(module)\n========================= [FAILED] Took 3.23 seconds ========================="`

Wiz-IO, 2019-10-07:
EC2x is not support yet ...

LaBus3, 2019-10-07:
> 
> 
> EC2x is not support yet ...

Coming from the YouTube videos showing EC25 programs, creating EC25 project also fails. Is it not yet supported?

Wiz-IO, 2019-10-07:
experiments, in progress ...

closed 2019-10-07 by LaBus3