Wiz-IO/platform-quectel #14: BG96 - Arduino Won't build

akinsony, 2019-10-16:
Hi there,

Thanks for the effort putting together this library for Quectel modules. I am trying to get the BG96 going on Arduino - Platform IO.

When I build I get the following:

> Executing task: C:\Users\root\.platformio\penv\Scripts\platformio.exe run <

Processing bg96 (platform: quectel; board: bg96; framework: arduino)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
?[32m<<<<<<<<<<<< QUECTEL BG96 MODULE 2018 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/quectel/bg96.html
PLATFORM: Quectel 2.0.3 #fbd7a6e > Quectel BG96 module
HARDWARE: ARM, MDM9206 1200MHz, 3.00MB RAM, 3.00MB Flash
DEBUG: Current (uart) On-board (uart)
PACKAGES: framework-quectel 44bdff7, toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
CORE bg96 SDK2 RO_BASE = 0x40000000 HEAP = 1048576
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 6 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pio\build\bg96\src\main.cpp.o
Compiling .pio\build\bg96\_arduino\IPAddress.cpp.o
Compiling .pio\build\bg96\_arduino\Print.cpp.o
Compiling .pio\build\bg96\_arduino\Stream.cpp.o
Compiling .pio\build\bg96\_arduino\WMath.cpp.o
Compiling .pio\build\bg96\_arduino\WString.cpp.o
Compiling .pio\build\bg96\_arduino\cbuf.cpp.o
Compiling .pio\build\bg96\_arduino\dtostrf.c.o
Compiling .pio\build\bg96\_arduino\new.cpp.o
Compiling .pio\build\bg96\_arduino\wiring_pulse.c.o
Compiling .pio\build\bg96\_arduino\wiring_shift.c.o
Compiling .pio\build\bg96\_core\HardwareSerial.cpp.o
Compiling .pio\build\bg96\_core\arduino_main.cpp.o
Compiling .pio\build\bg96\_core\ril.cpp.o
Compiling .pio\build\bg96\_core\std.c.o
Compiling .pio\build\bg96\_core\wiring.cpp.o
Archiving .pio\build\bg96\lib_arduino.a
Compiling .pio\build\bg96\_core\wiring_analog.c.o
Compiling .pio\build\bg96\_core\wiring_digital.c.o
Compiling .pio\build\bg96\_variant\variant.cpp.o
Compiling .pio\build\bg96\_variant\variant_pins.c.o
Compiling .pio\build\bg96\_threadx\interface.c.o
Archiving .pio\build\bg96\lib_core.a
Compiling .pio\build\bg96\_threadx\preamble.c.o
Compiling .pio\build\bg96\_threadx\qapi_src.c.o
Archiving .pio\build\bg96\lib_variant.a
Compiling .pio\build\bg96\_wizio\common.c.o
Compiling .pio\build\bg96\_wizio\dbg.c.o
Compiling .pio\build\bg96\_wizio\scanf.c.o
Archiving .pio\build\bg96\lib_threadx.a
Compiling .pio\build\bg96\_wizio\snprintf.c.o
Archiving .pio\build\bg96\lib_custom.a
Archiving .pio\build\bg96\lib_wizio.a
Linking .pio\build\bg96\program.elf
Checking size .pio\build\bg96\program.elf
Building .pio\build\bg96\program.bin
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  33.6% (used 1056336 bytes from 3145729 bytes)
PROGRAM: [          ]   0.4% (used 13213 bytes from 3145729 bytes)
ADD HEADER
*** [.pio\build\bg96\program.ini] TypeError : a bytes-like object is required, not 'str'
Traceback (most recent call last):
  File "C:\Users\root\.platformio\packages\tool-scons\script\..\engine\SCons\Action.py", line 1209, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "C:\Users\root\.platformio\platforms\quectel\builder\frameworks\arduino-bg96.py", line 24, in dev_header
    f.write("/datatx/program.bin")
TypeError: a bytes-like object is required, not 'str'

I will appreciate your assistance

Wiz-IO, 2019-10-16:
yep - thanks !
I make platform Python 3 compatible and have bugs yet
Today will update

akinsony, 2019-10-16:
> edit your files: arduino-bg96.py and threadx-bg96.py ... "wb" to "w+"
> b18b898
>
> or delete folder C:\Users\USER_NAME.platformio\platforms*quectel*
> and install platform again

Thanks, Georgi! I will update and retry

akinsony, 2019-10-17:
Hi Georgi,
My BG96 has 3 COM ports when connected via the USB connector. I am using VS Code and Platform IO. I am unable to program the board or get the Serial.println working on any of the ports. Am I missing something?
Cheers

Wiz-IO, 2019-10-17:
Hi, 
Arduino use hardware uarts...
https://github.com/Wiz-IO/framework-quectel/blob/master/arduino/variants/bg96/variant.cpp#L8
ThreadX not suport USB or I don know how... no examples

https://www.youtube.com/watch?v=DS5NEhEv1hA&t=1s
for Arduino is the same

akinsony, 2019-10-18:
This is an issue with this line

    print(str(Fore.BLUE) +  'Upload from Project folder ') #+ str(env.subst("$BUILD_DIR").replace('\\', '/')))

The concatenation is causing an issue and preventing successful building..

I have commented out #+ str(env.subst("$BUILD_DIR").replace('\\', '/')))

Also, do you know of a way that PlatformIO can directly upload the bin file to the BG96? Using the
QEFS_Explore works but it is just a pain to work with. It will be really neat if we can upload directly from PlatformIO

[image]

akinsony, 2019-10-18:
> Hi,
> Arduino use hardware uarts...
> https://github.com/Wiz-IO/framework-quectel/blob/master/arduino/variants/bg96/variant.cpp#L8
> ThreadX not suport USB or I don know how... no examples
> 
> https://www.youtube.com/watch?v=DS5NEhEv1hA&t=1s
> for Arduino is the same

Thanks mate!

I have finally got it to work using a RS232 to USB converter and pinning out Tx Rx and GND from the board. BTW I am using the SixFab BG96 for Raspberry Pi board.

Thanks for your swift help!

Wiz-IO, 2019-10-18:
Thanks - will check this line...
I did not succeed to reverse Qualcomm's protocol ( QEFS_Explore  ) ... will try again later

Wiz-IO, 2019-10-18:
https://github.com/Wiz-IO/platform-quectel/blob/master/builder/frameworks/arduino-bg96.py#L13

akinsony, 2019-10-18:
> Thanks - will check this line...
> I did not succeed to reverse Qualcomm's protocol ( QEFS_Explore ) ... will try again later

I think I have found the Quectel documentation for reverse-engineering the QEFS_Explore ...
https://os-iot.org/wp-content/uploads/2019/01/os-iot-bg96-guide.pdf

Wiz-IO, 2019-10-18:
3 more commands left :)

[image]

akinsony, 2019-10-18:
Awesome!!! Hopefully you can complete it soon

akinsony, 2019-10-18:
That is great news!!! I will wait for your got commit. This will make the process faster

Wiz-IO, 2019-10-18:
I will enable  option soon...

update your port
unmark lines 9 and 13 and test
https://github.com/Wiz-IO/platform-quectel/blob/master/builder/frameworks/arduino-bg96.py#L9

project INI add
;upload_port = COM8
;monitor_port = COM8
;monitor_speed = 115200

akinsony, 2019-10-18:
Testing...

akinsony, 2019-10-18:
I am getting ...
Checking size .pio\build\bg96\program.elf
Memory Usage -> http://bit.ly/pio-memory-usage
DATA:    [===       ]  33.7% (used 1058768 bytes from 3145729 bytes)
PROGRAM: [          ]   0.9% (used 28569 bytes from 3145729 bytes)
Configuring upload protocol...
Looking for upload port...
Use manually specified: COM31
Uploading: program
CONNECTING TO DM-PORT <
ERROR: read data
*** [upload] Explicit exit, status 1
====================================================================== [FAILED] Took 1.49 seconds ======================================================================

Trying to find the problem

Wiz-IO, 2019-10-18:
upload_port = COM8 ; **DM port**

akinsony, 2019-10-18:
I have commented out Line 180 and 181 in the QDL.py and it seems to work

        #ASSERT(len(rs) == len(rx), 'read len')
       # ASSERT(rs == rx, 'read data')

Wiz-IO, 2019-10-18:
lines in connect() function is unknown for me - it is like as enable EFS mode
https://github.com/Wiz-IO/platform-quectel/blob/master/builder/frameworks/QDL.py#L184

closed 2019-10-23 by Wiz-IO