Wiz-IO/platform-quectel #15: BC26 module cannot upload @ win VScode

D-zz, 2019-10-22:
Board： BCB6-TE-B

Configuring upload protocol...
Looking for upload port...
Use manually specified: COM3
Uploading: program
Waiting module for POWER-ON or RESET <===========================> DONE
Starting <========
ERROR: EPP Answer
*** [upload] Explicit exit, status 2

Wiz-IO, 2019-10-22:
more info? module firmware? platform version?
[image]

D-zz, 2019-10-22:
@Wiz-IO  Thank you very much. More info as below：

[image]

Wiz-IO, 2019-10-22:
try edit this value
https://github.com/Wiz-IO/platform-quectel/blob/master/builder/frameworks/MT2625.py#L193

Wiz-IO, 2019-10-22:
or enable DEBUG
https://github.com/Wiz-IO/platform-quectel/blob/master/builder/frameworks/MT2625.py#L61

D-zz, 2019-10-22:
My board is BC26-TE-B，it is same of BC66-TE-B？

Wiz-IO, 2019-10-22:
but maybe chip-set ID is other

Wiz-IO, 2019-10-22:
are you update BC26 with BC66 firmware?

D-zz, 2019-10-22:
@Wiz-IO  Yes ，i update BC26 with BC66 firmware
[image]

D-zz, 2019-10-22:
what's the error of “EPP Answer” means ?

D-zz, 2019-10-22:
when i use BC26, is same problem of use BC66

[image]

Wiz-IO, 2019-10-22:
“EPP Answer” means: no answer from first bootloader

closed 2019-11-08 by Wiz-IO

maximevince, 2021-03-31:
If anyone is still wondering, I ran into the same issue, and it can be solved as follows:
- the BC66 needs some more time after booting, before receiving the expected `EPP` reply. (Almost 500us)

This can be seen in the capture below:
[image]

It can be solved by modifying ~/.platformio/platforms/quectel/builder/frameworks/MT2625.py by adding `time.sleep(0.5)` or similar to line 384:

```
diff --git a/builder/frameworks/MT2625.py b/builder/frameworks/MT2625.py
index e9215ac..898a8cc 100644
--- a/builder/frameworks/MT2625.py
+++ b/builder/frameworks/MT2625.py
@@ -381,6 +381,8 @@ class MT2625:
             self.s.write(data[:1024])                        
             data = data[1024:]  
             PB_STEP()   
+        # Give the chip some time
+        time.sleep(0.5)
         ASSERT( self.cmd("", 10) == b"\x1D\x3D\0\0\0\0\0\0\0\0", "EPP Answer" ) 
         self.uart_speed_max()           
         # DA_2
```

Wiz-IO, 2021-03-31:
Thanks @maximevince

maximevince, 2021-03-31:
Thanks @Wiz-IO for your platformio framework. I got a custom application running on a BC66 in no time. Awesome!