Wiz-IO/platform-quectel #16: OpenCPU FOTA Package format

glaucopgomes, 2019-10-26:
Do you have any documentation or knowledge of the OpenCPU FOTA Package format?

I'd like to automate the creation of the update package after build the application.

Wiz-IO, 2019-10-26:
Hi,
ask Quectel....
FOTA is normal application plus header
DFOTA is differencial between two applications - old and new

closed 2019-10-31 by Wiz-IO

rurban, 2021-05-16:
Apparently it is EC20 as reverse engineered here:
https://penthertz.com/blog/mobile-iot-modules-FOTA-backdooring-at-scale.html
and described here: https://osmocom.org/projects/quectel-modems/wiki/EC20_DFOTA

This is only usable for the base FW, not for your application. 
For your application FW you need to implement it by yourself.
There I would just use jojodiff. Would be interesting to be able to use the live App FW data as base for jptch.