Wiz-IO/platform-quectel #20: Cannot compile Opencpu C++ project

ASL07, 2020-02-26:
Hi,

I am having trouble compiling a c++ project

I created a pio project with this platformio.ini:
```
[env:bc66]
platform = quectel
board = bc66
framework = opencpu
```

When I compile using "pio run" , a new main.c file is created. 
As mentioned here https://github.com/Wiz-IO/platform-quectel/wiki/SDK-BC66

> Create New OpenCPU Project, PIO will make basic C template...
> Rename main.c to main.cpp

I have renamed main.c to main.cpp

When I try to compile again, a new main.c file is created
How do I use opencpu framework and c++?

Thanks for your help

Wiz-IO, 2020-02-26:
Hi
Please, read the first line
https://github.com/Wiz-IO/platform-quectel/wiki/SDK-BC66#new-version-comming-soon

ASL07, 2020-02-26:
Do you mean this?

> New version comming soon...

So, we have to wait for the new version?

Wiz-IO, 2020-02-26:
mean: new version is not ready

closed 2020-02-26 by Wiz-IO

ASL07, 2020-02-26:
So, with the current version, we can't compile cpp code?
When is the new version going to be available?
Thanks

Wiz-IO, 2020-02-26:
current version not support "clear" cpp
you can use Arduino as cpp

ASL07, 2020-02-26:
Ok
I might stick with C or Arduino for the moment but it will be great if you could provide a timeline for the new version release.
Maybe the docs should be updated to mention that c++ is not supported yet

ASL07, 2020-03-11:
Is this fixed in the latest published release?

Wiz-IO, 2020-03-11:
C++?
https://github.com/Wiz-IO/platformio-quectel-examples/blob/master/BC66_SDK/CPP/BC66_CPP_HELLO/src/main.cpp