Wiz-IO/platform-quectel #21: upload_port platform-quectel linux

j054n, 2020-03-03:
Hi Mr. Angelov,

I have just used the platform-quectel with "success" in Windows, despite of several hangs with uart ports.
I would prefer to use in Linux but configuring in platformio.ini as follows, with regular usb-serial ports, I only can monitor but I got error in upload.
Is possible it could be due to QDL.py?  Any solution?
Thanks in advance.

[env:bg96]
platform = quectel
board = bg96
framework = threadx
; select DM COM Port
upload_port = /dev/ttyUSB2
; select module uart
monitor_port = /dev/ttyUSB4
monitor_speed = 115200

Wiz-IO, 2020-03-03:
Hi
are you sure in your Linux drivers... must be selected DM port as windows

j054n, 2020-03-03:
Any port of QDL script for Linux?

Wiz-IO, 2020-03-03:
is a Python script, can work of all platforms

closed 2020-03-12 by Wiz-IO