Wiz-IO/platform-quectel #22: opencpu undefined reference to _sbrk

ASL07, 2020-03-11:
It seems that syscalls.c is not being linked
This is the error I am getting. I am not using dynamic memory allocation

```
<<<<<<<<<<<< OLIMEX NB-IOT-DEVKIT 2018 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/quectel/Olimex-NB-IoT-DevKit.html
PLATFORM: Quectel 2.0.14 #5dbec3f > Olimex NB-IoT-DevKit
HARDWARE: ARM, MT2625 78MHz, 97.66KB RAM, 195.31KB Flash
DEBUG: Current (uart) On-board (uart)
PACKAGES: 
 - framework-quectel b91f56c 
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio/build/bc66/src/as_main.o
Linking .pio/build/bc66/program.elf
/Users/developer/.platformio/packages/toolchain-gccarmnoneeabi/bin/../lib/gcc/arm-none-eabi/7.2.1/../../../../arm-none-eabi/lib/thumb/v7e-m/fpv4-sp/hard/libc.a(lib_a-sbrkr.o): In function `_sbrk_r':
sbrkr.c:(.text._sbrk_r+0xc): undefined reference to `_sbrk'
collect2: error: ld returned 1 exit status
*** [.pio/build/bc66/program.elf] Error 1
```

I have seen a fix here https://github.com/svaarala/duktape/issues/1941 that consists on adding this to a source file:
```
void *_sbrk(s32 inc)
{
	APP_DEBUG("_sbrk called.\r\n");
    while(1);
	return NULL;
}
```
however there must be a better solution. 

Please let me know,
Thanks

Wiz-IO, 2020-03-11:
you use old PLATFORM: Quectel 2.0.14
there is new Version: 2.1.01
https://github.com/Wiz-IO/platform-quectel#quectel-development-platform-for-platformio

sbrk included in project
https://github.com/Wiz-IO/platformio-quectel-examples/blob/master/BC66_SDK/CPP/BC66_CPP_HELLO/config/custom_sys_cfg.c#L36

Wiz-IO, 2020-03-11:
[image]

ASL07, 2020-03-11:
I have updated the platform to the latest. The are several issues:

1. URC_SOCKET_RCV_DATA is not defined anymore
`src/as_main.c:193:14: error: 'URC_SOCKET_RECV_DATA' undeclared (first use in this function); did you mean 'URC_SOCKET_CLOSE'?`

2. It seems that the app is not loading. I have programmed the example_gpio.c and I get no output in the usb serial port. And also the Led doesn't change state.
I have this in my custom_sys_cfg.h
```
static const ST_AppEnable appEnableCfg = {
    APP_ENABLE
};
```

3. Weird compilation warnings about non unicode characters on these files
```
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from /Users/developer/.platformio/packages/framework-quectel/opencpu/bc66/SDK15/ril/inc/ril_dfota.h file
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from /Users/developer/.platformio/packages/framework-quectel/opencpu/bc66/SDK15/ril/inc/ril_socket.h file
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from /Users/developer/.platformio/packages/framework-quectel/opencpu/bc66/SDK15/ril/src/ril_network.c file
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from /Users/developer/.platformio/packages/framework-quectel/opencpu/bc66/SDK15/ril/src/ril_sim.c file
Unicode decode error has occurred, please remove invalid (non-ASCII or non-UTF8) characters from /Users/developer/.platformio/packages/framework-quectel/opencpu/bc66/SDK15/ril/src/ril_system.c file
```

Wiz-IO, 2020-03-11:
ah... just read wiki
https://github.com/Wiz-IO/platform-quectel/wiki/SDK-BC66#opencpu

**By default some RIL functions is disabled, see your project/custom_feature_def.h**

ASL07, 2020-03-11:
I was compiling this project with the previous version of the platform, and the app was running fine.
Once I upgraded to the latest version, then I started getting this errors. I have not touched my custom_feature_def.h, so that should not be the problem

Here is the content of custom_feature_def.h just in case
```
#ifndef __CUSTOM_FEATURE_DEF_H__
#define __CUSTOM_FEATURE_DEF_H__

/************************************************************************
 * RIL Function on/off
 ************************************************************************/
#define __OCPU_RIL_SUPPORT__
#define __OCPU_RIL_SOCKET_SUPPORT__

// #define __OCPU_RIL_LWM2M_SUPPORT__
// #define __OCPU_RIL_MQTT_SUPPORT__
// #define __OCPU_RIL_DFOTA_SUPPORT__

#endif  //__CUSTOM_FEATURE_DEF_H__

```

Wiz-IO, 2020-03-11:
uncomment
#define __OCPU_RIL_LWM2M_SUPPORT__
#define __OCPU_RIL_MQTT_SUPPORT__
#define __OCPU_RIL_DFOTA_SUPPORT__

ASL07, 2020-03-11:
I've already done it and it doesn't make any difference.
Why should I uncomment them? I don't need support for those

Wiz-IO, 2020-03-11:
your example is from old OpenCPU SDK
new OpenCPU v1.5 not use URC_SOCKET_RECV_DATA

there is new Quectel examples
https://github.com/Wiz-IO/platformio-quectel-examples/tree/master/BC66_SDK/OpenCPU

ASL07, 2020-03-11:
I get this problem when programming the new example

> It seems that the app is not loading. I have programmed the example_gpio.c and I get no output in the usb serial port. And also the Led doesn't change state.

This other problem  unrelated to the previous one. I get this everytime I try to use URC_SOCKET_RECV_DATA.  I have checked the new tcp exampe, so I should be using `Ql_Socket_Recv_Register()` instead in the new SDK?
> URC_SOCKET_RCV_DATA is not defined anymore
> src/as_main.c:193:14: error: 'URC_SOCKET_RECV_DATA' undeclared (first use in this function); did you mean 'URC_SOCKET_CLOSE'?

Thanks

Wiz-IO, 2020-03-11:
are you read me?
URC_SOCKET_RECV_DATA is not used in OpenCPU v1.5

Wiz-IO, 2020-03-11:
old OpenCPU not have API for sockets and use RIL ( AT commands ) for this
New v1.5 have API ( direct call to lwip core functions )
explore https://github.com/Wiz-IO/framework-quectel/blob/master/opencpu/bc66/SDK15/include/ql_socket.h

or ask Quectel: WHY?
I just follow OpenCPU SDK

ASL07, 2020-03-11:
Thank you for the clarification

Please, understand that most of us don't have access to the OpenCPU documentation and Quectel does not provide support for OpenCPU unless you place a really big order...
It is difficult for us to determine what the changes are from version to version without having access to a changelog

Do you have any idea regarding what can be causing the application not to load? Regards

Wiz-IO, 2020-03-11:
Hi, I not have Quectel support too...
email me

Wiz-IO, 2020-03-11:
btw: update your module with last firmware BC66NBR01A10

Wiz-IO, 2020-03-11:
I test platform with original Quectel board and Olimex board and with last two firmwares
BC66NBR01A07 and BC66NBR01A10 
yesterday I update my Olimex to BC66NBR01A10 and make some tests
I can not see problems with uploading
btw: BC66 uploader script not have changes

ASL07, 2020-03-12:
I have upgraded the FW and now it works. That was the problem

closed 2020-03-12 by ASL07