Wiz-IO/platform-quectel #23: EC2x unofficial release?

koblibri, 2020-03-17:
Hi there,

thanks for the effort in building this framework.
I'm trying to connect an ESP32 AWS over LTE with the EC25 and coming from your YouTube video, I can see that a cellular connection is possible. Reading the other issues and descriptions, I understand that the EC25 Arduino Framework is currently not supported fully. 
Nevertheless I wanted to ask if you could share the code for platform initialization so I could fiddle around with it and try to do a proof-of-concept. I would not care if there might be bugs or not everything is working as intended.

Any help would be appreciated!

Wiz-IO, 2020-03-17:
Hi
 _if you could share the code for platform initialization...._ **soon** ... I need "free" time
About full EC2x SDK - you need to ask Quectel

koblibri, 2020-03-18:
Alright, it was worth a shot. Thanks anyway!

closed 2020-03-18 by koblibri