Wiz-IO/platform-quectel #24: BG96 blocks COM ports after reset and hang up

j054n, 2020-03-19:
Previously I've been using your Quectel platform with PlatformIO in UMTS & LTE EVB Kit with BG96, uploading from VSCode and QEFS_Explorer, but now COM ports are blocked inesperately:
* I reset by button and platform hang up, like going sleep, and now it’s not possible flash again, it only appears 1 USB port (DM port) but it’s not possible to flash again, QEFS explorer hangs and exit.
* I have tried again long reset, three times reset and I with in device manager likes appear 3 ports and inmediately disappear and only one remains (DM) 
Any trick to reset completely flash or access by DM port? I’m desperate.
Thanks in advance.
PD: I ask you directly becase I can't post by now in Quectel forum

Wiz-IO, 2020-03-19:
Hi
at module board ( the small board ) exist control point for usb-boot
connect resistor 1k VDD_EX ( 1.8v ) to point and power ON module
then you will see new usb-com port, use it and reflash full

j054n, 2020-03-19:
Thanks, I'll try to do it, but I don't have FW to refresh full
Are these cases usual? I understand you had the same problem

Wiz-IO, 2020-03-19:
ask from Quectel firmware 
**BG96MAR02A09M1G_01.011.01.011** (tested with PlatformIO)
and reflash module

j054n, 2020-03-24:
Finally, I got reflash BG96 module with firmware that support can facilitate to me:
BG96MAR02A08M1G_01.013.01.013
It's previous than you suggest, can it supose any problem like previous hang up?

With respect to other question, to upload in PlatformIO I have find the problem to do it in Linux:
 [image]

Wiz-IO, 2020-03-24:
about firmwares? module support all firmwares
your firmware is OK...

About Linux... I not have Nix platform and I can not test ( I test only compilator via Virtual )
About Uploader, you need USB driver that support DM com port, you need to contact Quectel and ask for driver or look here :
https://github.com/ngohaibac/Quectel_USB_Serial_Driver
or contact with HaiBac Ngo ngohaibac

closed 2020-04-08 by Wiz-IO

jiangshao6, 2025-02-08:
你们是怎么把UMTS&LTE-EVB-KIT这平台用起来的，需要配置什么参数吗？