Wiz-IO/platform-quectel #25: Flashing using PlatformIO fails for BC66-NA

bart-iot, 2020-04-02:
Hi there, 

I am using your platform for developing openCPU code for a BC66-NA module. It works great but flashing using PlatrformIO fails. Flashing the binary with IoT flash tool works fine. 
PlatformIO reports this:

> Use manually specified: COM9
> Uploading: program
> Waiting module for POWER-ON or RESET <=========================> DONE
> Starting <========
> ERROR: EPP Answer
> *** [upload] Explicit exit, status 2

Monitoring using PlatformIO works fine. 

Any ideas what can cause this?

Wiz-IO, 2020-04-02:
Hi
USB driver...

Wiz-IO, 2020-04-02:
BC66-**NA** ? I not have NA module for test

bart-iot, 2020-04-02:
Thanx for your reply. I also have a BC66NB evaluation kit, that has the same problem with flashing. I installed the drivers according the documentation, but problem stays.
I think there is not really a big difference between BC66NB or BC66-NA .. at least the delivered SDK is identical

Wiz-IO, 2020-04-02:
IoT Tool work? ( original flasher )

bart-iot, 2020-04-02:
Yes that works fine.

Wiz-IO, 2020-04-02:
no idea .... use IoT Tool ... app bin file is in project build folder

bart-iot, 2020-04-03:
Yes I already did that, so will keep doing that. Thanx. :)

closed 2020-04-03 by bart-iot

amaljayaprakash, 2021-03-03:
i am also getting the same error.. but iot flash tool works file
[image]

Wiz-IO, 2021-03-03:
install drivers or use IoT Falsh Tool

BionicSystemSolutions, 2021-03-19:
Can you link to a download page for this flash tool?

Wiz-IO, 2021-03-19:
https://github.com/Wiz-IO/LIB/blob/master/quectel/IoT_FlashTool_v2.7.3.zip

maximevince, 2021-03-31:
I ran into a similar issue (using BC66), see https://github.com/Wiz-IO/platform-quectel/issues/15#issuecomment-810963028 for a possible solution.