Wiz-IO/platform-quectel #27: Error: Multiple ways to build the same target were specified

1iveowl, 2020-04-09:
I trying to get up and running, but I'm stuck trying to build the `bc66_cpp_hello` example:

[image]

I have verified that the correct Quectel platform is installed: 
[image]

A hit for how to proceed would be great appreciated.

Wiz-IO, 2020-04-09:
Hi
you have two main files in project

Wiz-IO, 2020-04-09:
[image]

Wiz-IO, 2020-04-09:
[image]

1iveowl, 2020-04-09:
Ahh, of course. I just downloaded the example. Didn't look too close at the code. 

Not sure why the example has two mains? Anyhow I commented out the C++ one.

Wiz-IO, 2020-04-09:
if you want to do experiments with C++
create new OpenCPU project, then rename main.c to main.cpp then follow the example
https://github.com/Wiz-IO/platformio-quectel-examples/blob/master/BC66_SDK/CPP/BC66_CPP_HELLO/src/main.cpp

sorry, is only one example... I need time to make other...

1iveowl, 2020-04-09:
> sorry, is only one example... I need time to make other...

No problem. I should have looked at the code.

I've tried with both c and c++. I can compile both however with `main.cpp` I get a squiggly line under `#include <os_wizio.h>`, and with `main.c` I get squiggly lines under all the includes except `stdint.h` and `stddef.h`. Why is that? As I said I can compile them just fine.

Wiz-IO, 2020-04-09:
this folder <**wizio**> is my drivers and helpers for Arduino and C++ projects
named is ... os_wizio.h 

this drivers can not be used in normal OpenCPU project.... is not included in search path
for OpenCPU you need to follow Quectel SDK
**include** and **ril** folder
https://github.com/Wiz-IO/framework-quectel/tree/master/opencpu/bc66/SDK15/include

1iveowl, 2020-04-09:
When trying to compile the tcp_http example I get: 
> fatal error: nbClient.h: No such file or directory

Am I missing another lib?

Wiz-IO, 2020-04-09:
for last platform version examples is there
https://github.com/Wiz-IO/platformio-quectel-examples/tree/master/BC66_SDK

wiki is:
https://github.com/Wiz-IO/platform-quectel/wiki/SDK-BC66

closed 2020-04-13 by Wiz-IO