Wiz-IO/platform-quectel #29: Python based flashtool

ajaybhargav, 2020-06-03:
Hi @Wiz-IO Did you get python based flashtool working? I am also looking for a cross-platform flasher for Mediatek chipset. Maybe we can collaborate to make it work?

Wiz-IO, 2020-06-03:
Hi,
what chipset?

MT2625 must work on Win/Lin/Mac ( just need usb drivers )
https://github.com/Wiz-IO/platform-quectel/blob/master/builder/frameworks/MT2625.py

ajaybhargav, 2020-06-03:
MT2503/MT6261

Wiz-IO, 2020-06-03:
https://github.com/Wiz-IO/platform-quectel/blob/master/builder/frameworks/MT2625.py

probonopd, 2020-06-03:
Does this mean all MT2503/MT6261 based modules can work?

Wiz-IO, 2020-06-03:
yep, the boot loader is one
test ... or **spy** COM port for reverse protocol
I not have info from Mediatek/Quectel

ajaybhargav, 2020-06-03:
wow thats a lot of work i believe. Will it support flashing via usb? for mediatek usb is just a name they are creating a usbcom port and doing usual uart stuff.

Wiz-IO, 2020-06-03:
USB direct?
or UART ( Quectel bootloader is uart based )

ajaybhargav, 2020-06-03:
I am not using MC60. its a different module based on MT2503 with USB interface exposed. When connected it creates USB com port for flashing, rest everything looks similar to uart however i may be wrong. i will put a sniffing tool to see what's going on. which tool did you use for uart sniffing?

Wiz-IO, 2020-06-03:
the princip:

- enter boot mode: send 0xA0 .... A0, read 0x5F  - boot mode OK ( maybe diff numbers )
- read SoC info
- upload boot part 1 ( from MTK_AllOne_DA.bin )
- upload boot part 2 ( from MTK_AllOne_DA.bin ) this parts is based of SoC ID
- upload firmware, format fat.... any commands
 :)

ajaybhargav, 2020-06-04:
I am going to test it today :) lets see. Which tool did you use to monitor (spy) serial port?
Do you have any idea how hwids in platformio can be used to detect dynamic serial port?

Wiz-IO, 2020-06-04:
I use:
https://www.hhdsoftware.com/

ajaybhargav, 2020-06-04:
I am using the same but It is not able to capture USB port of module, serial works fine.

ajaybhargav, 2020-06-04:
I am able to capture both USB and UART... here are the application flashing logs.

[mt2503_data.zip](https://github.com/Wiz-IO/platform-quectel/files/4728671/mt2503_data.zip)

Wiz-IO, 2020-06-04:
is this USB-HID or is USB-UART?

Wiz-IO, 2020-06-04:
the first part: enter boot-mode is the same
```py
    def connect(self, timeout = 9.0):
        self.s.timeout = 0.02
        c = 0
        PB_BEGIN()
        while True:
            if c % 10 == 0: PB_STEP()
            c += 1
            self.s.write( b"\xA0" )      
            if self.s.read(1) == b"\x5F":        
                self.s.write(b"\x0A\x50\x05")
                r = self.s.read(3)
                if r == b"\xF5\xAF\xFA":  
                    break
                else: ERROR("BOOT")
            timeout -= self.s.timeout
            if timeout < 0:
                ERROR("Timeout") 
```

ajaybhargav, 2020-06-04:
> 
> 
> is this USB-HID or is USB-UART?

Its USB-UART (in log it says Device: MediaTek USB Port (COM12) (MT6227 phone))

yes first part is same.. then it flashtool go crazy with some kind of settings on external memory interface (is this needed?)
0xA0050000 <- base address for external memory interface

```
A0 05 01 10 = 00 00 00 00
A0 05 00 60 = 00 00 10 01
A0 05 00 70 = 00 00 00 11
A0 05 00 88 = 00 00 00 01
A0 05 01 30 = 00 01 00 01
A0 05 01 38 = 00 01 00 01
A0 05 01 50 = 00 00 50 00
A0 05 01 58 = 00 00 10 01
A0 05 00 90 = 00 00 00 00

-- Loop till 00 00 00 00
A0 05 00 D0 = 1F 1F 1F 1F
A0 05 00 D8 = 1F 1F 1F 1F
A0 05 00 E0 = 00 00 00 00
A0 05 03 00 = 00 00 00 00
A0 05 03 00 = 00 00 00 00
A0 05 03 08 = 20 00 00 3F
A0 05 03 00 = A5 5A 00 00
A0 05 03 00 = A5 5A 03 25
d1 A0 05 03 18 00 00 00 01> -- dummy read?
d1 A0 05 03 18 00 00 00 01> -- dummy read? both read returns same value
A0 05 03 00 = 00 00 00 00
```

scanning manually though that loop took a lot of time but I am almost lost after that :)

Wiz-IO, 2020-06-04:
your SoC is 90% as my python script, bold is differents

```
Enter Boot: _same_
Read SoC Info need to get DA parts[2] from MTK_AllOne_DA.bin
        
>>> 80 00 00 00 00 00 00 01 
<<< 80 00 00 00 00 00 00 01 [cb 01] BB_CPU_HW = CB01

>>> 80 00 00 04 00 00 00 01
<<< 80 00 00 04 00 00 00 01 [00 01] BB_CPU_SW = 0001

>>> 80 00 00 08 00 00 00 01
<<< 80 00 00 08 00 00 00 01 [62 61] BB_CPU_ID = 6261

>>> 80 00 00 0c 00 00 00 01    
<<< 80 00 00 0c 00 00 00 01 [80 00] BB_CPU_SB = 8000  

**Prepare DA ( download agent )**

SEND_DA_1: _same_
SEND_DA_2: _same_

CMD_JUMP_DA: _same_
**FLASH: Init** 
UPLOAD FIRMWARE: _same_
```

ajaybhargav, 2020-06-04:
Thanks @Wiz-IO MT2503 is basically MT6261 + MT3333 in single SoC. so I believe most things should be same.
Can you explain
> Prepare DA DIFFERENT

And with "FLASH: Init DIFFERENT" you meant the start address etc. correct?

Wiz-IO, 2020-06-04:
Prepare DA is set some params before uploadind DA (Download Agent)
after upload - CMD_JUMP_DA( start DA )  
then get/set FLASH info ( for flash as chip )
then clear/upload flash ... and other commands

ajaybhargav, 2020-06-04:
It would be helpful if you can reference line number from [MT6261.py](https://github.com/Wiz-IO/platform-quectel/blob/master/builder/frameworks/MT6261.py) script. I can then try to modify it.

Did you check USB logs? they do look similar to UART but some extra stuff going on which I do not understand. I mentioned it here https://github.com/Wiz-IO/platform-quectel/issues/29#issuecomment-638739457

Wiz-IO, 2020-06-04:
at the beginig - try separate SEND-RECEIVE data from spy-log
and write notes to "reverse" protocol
then is easy :)

ajaybhargav, 2020-06-04:
Thanks, I will try that 😄

ajaybhargav, 2020-06-05:
Hi Georgi (@Wiz-IO)
I tested standalone MT6261.py today. And with just modification to application address (DEVICE), everything worked perfectly. No change made to DA, as I believed DA must be same for all MTK devices. I tested over uart as of now no USB.

I had to comment few things in the source:
Below part I had to comment completely as the MMM check always fail.
 ``` python
        if check == True:
            if app_data[:3] != "MMM":
                ERROR("APP: MMM")
            if app_data[8:17] != "FILE_INFO":
                ERROR("APP: FILE_INFO")
```

I am getting error "APP: MMM" even though file starts with MMM still it throw error.

I also had to comment and change the following:
```python
def crc_word(self, data, chs=0):
        for i in xrange(0, len(data), 1):
            chs += data[i] & 0xFF #ord(data[i])
        return chs & 0xFFFF
```

Error with ord says:
```
Exception has occurred: TypeError
ord() expected string of length 1, but int found
```
So far so good :)

closed 2020-06-25 by ajaybhargav