Wiz-IO/platform-quectel #33: Getting uS resolution on timers or delay function

ASL07, 2020-06-15:
Hello,
I need to interface OpenCPU in BC66 with a HC-SR04T ultrasonic sensor.
The sensor needs a trigger pulse of 10uS and will return an echo pulse of a certain uS duration, which has to be measured by the mcu.

[image]

Is it possible to have uS resolution on OpenCPU timers (I only see millisecond resolution in the examples), or a delay_us() function?

Thanks

ASL07, 2020-06-15:
I found `s32 Ql_Timer_Register_us(u32 timerId, Callback_Timer_OnTimer callback_onTimer, void* param);`

Will try that

Wiz-IO, 2020-06-15:
yep, but GPIO is slow...
GSM modules is not MCU .... unknow SoC, closed RTOS & API, watchdog... ets
use or make smart sensors on SPI, I2C, UART...

ASL07, 2020-06-17:
Hi @Wiz-IO 

Is there any api function similar to Arduino micros() for BC66 (using OpenCPU, not Arduino)?

I have seen this macro in os_wizio.h:
`#define HAL_MILLIS() (1000L * *((volatile uint32_t *)0xA2130118) >> 15) /* rollover at 0x20000 ms */`

and I tried to modify it to report micros instead of millis, like so:
`#define micros() (*((volatile uint32_t*) 0xA2130118) >> 15)`

However, micros() is always returning 1, am I doing something wrong?

Wiz-IO, 2020-06-17:
if I find free running us timer will write
if you want to use "local" get_micro() - there is a solution...

Wiz-IO, 2020-06-17:
for local use 
test....
https://github.com/Wiz-IO/LIB/blob/master/NOTES/micro_bc66.c#L57

Wiz-IO, 2020-06-17:
or check default cortex-m systick register ( I dont know freq )
```
#define ST_CURRENT   (*((volatile unsigned long *)0xE000E018))
a = ST_CURRENT ;
```

ASL07, 2020-06-18:
> for local use
> test....
> https://github.com/Wiz-IO/LIB/blob/master/NOTES/micro_bc66.c#L57

This works! Thank you
I can get relatively accurate values from the sensor now

Btw, how do you stop this timer once you don't need it anymore?

Wiz-IO, 2020-06-18:
((GPT_REGISTER_T *)0xA21300D0)->gpt->GPT_CON_UNION.GPT_CON = 0x0

ASL07, 2020-06-18:
Thank you for the support

closed 2020-06-18 by ASL07