Wiz-IO/platform-quectel #34: Ql_RIL_Initialize() blocks forever when using several tasks

ASL07, 2020-06-23:
Hi,

It seems that when using more than one task, Ql_RIL_Initialize() never returns and blocks the task from which it was called forever. 

I have uploaded a repro from which the error can be reproduced here:
https://github.com/ASL07/OpenCPURilInitHangMultitask

Please follow instructions on the README

Has anyone experienced anything similar?

Thanks

Wiz-IO, 2020-06-23:
**I am sorry but I am not Quectel FAE...**

BTW: your task config is wrong
https://github.com/ASL07/OpenCPURilInitHangMultitask/blob/master/config/custom_task_cfg.h#L53

first is the main task
next have two reserved tasks for RIL interface... and Ql_RIL_Initialize() block
next is user tasks
example:
https://github.com/Wiz-IO/framework-quectel/blob/master/templates/bc66/arduino_task_cfg.h#L5

ASL07, 2020-06-23:
Thank you, that was the problem to fix it.
It isn't mentioned anywhere that the order of tasks has to be strictly this:

```c
TASK_ITEM(proc_main_task,       main_task_id,   10*1024, DEFAULT_VALUE1, DEFAULT_VALUE2)
TASK_ITEM(proc_reserved1,       reserved1_id,    5*1024, DEFAULT_VALUE1, DEFAULT_VALUE2)
TASK_ITEM(proc_reserved2,       reserved2_id,    5*1024, DEFAULT_VALUE1, DEFAULT_VALUE2)

// User tasks must be defined after proc_reserved2
TASK_ITEM(user_task,        user_task_id,    10*1024, DEFAULT_VALUE1, DEFAULT_VALUE2)
```

closed 2020-06-23 by ASL07