Wiz-IO/platform-quectel #35: Request: Add support for Bg77 (BG95/BG300)

MrZANE42, 2020-06-26:
Hi,
The mentioned modules are very interesting since they are the latest low power very compact modules. They are also based on a never qualcom chipset so I don't know how easy it would be for you to support them but at least there is a quecopen SDK available (maybe still in Beta, not sure)
Tia
Jimmy

Wiz-IO, 2020-06-26:
BG95 is in test stage
https://www.youtube.com/watch?v=1vF9Lb7n_-k

for BG77 I not have SDK or they don't give it to me...

MrZANE42, 2020-06-26:
BG77 uses the same chipset as the BG95, it just lacks the EGPRS.
But of course there could be software differences that makes them
incompatible from a platform view.
Too bad you can't get the SDK, I guess it's probably not ready yet.
I do have a quecopen firmware for the BG77
(BG77LAR02A02_BETA1202A_01.003.01.003_BETA1202A) but I'm not allowed to
share it (under NDA)
But if you need anything tested on the BG77 just tell me and I'll see what
I can do.

Den fre 26 juni 2020 kl 08:03 skrev Georgi Angelov <notifications@github.com

Wiz-IO, 2020-06-26:
ThreadX struct is the same as BG95, maybe have difference for QAPI system calls
Will wait Quectel for stable firmware

closed 2021-03-30 by Wiz-IO

MrZANE42, 2021-03-30:
Does the closing of this issue mean that you won't add BG95/BG77, or that you are ready to add it and just haven't pushed it yet ;-)
Hopping for the former