Wiz-IO/platform-quectel #38: "ERROR: Boot timeout" on Mac OS when uploading through platformio.

OlineStaerke, 2020-11-21:
Hi, Im getting the error Boot Timeout, when I upload and click the start button on the board. 

Is it possible to fix? 

This is what I have : 

[env:bc66-te-b]

platform = quectel
board = bc66-te-b
framework = arduino
board_build.firmware = BC66NBR01A10
upload_port = /dev/tty.SLAB_USBtoUART3
monitor_port = /dev/tty.SLAB_USBtoUART3
monitor_speed = 9600

Thank you

Wiz-IO, 2020-11-21:
Hi
your problem is USB drivers
contact with your Quectel support and ask How to intsll drivers
I not have Mac - cant test

OlineStaerke, 2020-11-21:
Hi thanks, do you know which driver?

OlineStaerke, 2020-11-21:
I changed drivers, and now I get the following error:

Waiting module for POWER-ON or RESET <=========================> DONE
Starting <========
ERROR: EPP Answer
*** [upload] Explicit exit, status 2
====================================================================== [FAILED] Took 5.96 seconds ======================================================================
The terminal process "pio 'run', '--target', 'upload', '--environment', 'bc66'" terminated with exit code: 1.

Wiz-IO, 2020-11-21:
no idea about Mac drivers... I just know it works

closed 2020-12-17 by Wiz-IO