Wiz-IO/platform-quectel #40: Bug in math.h in toolchain

aakash4895, 2020-11-24:
Whenever I use sqrt function from math.h library QFlash shows this error
[image]

But when I the Quectel sdk which uses a different toolchain there is no such errors

Wiz-IO, 2020-11-24:
Hi
M66 and MC60 are 95% identical...
seem as Quectel uploader issue.... no idea for now

screenshots - same example - changed only module...

[image]
[image]

Wiz-IO, 2020-11-24:
[image]

aakash4895, 2020-11-24:
I solved the problem by using the toolchain from Sourcery Codebench Lite given by Quectel in platform io
[image]

[image]

[image]

aakash4895, 2020-11-24:
[image]

Wiz-IO, 2020-11-24:
Nice
I will ask Quectel

IF you need any compiler keys ( compilers is different ) edit here
https://github.com/Wiz-IO/platform-quectel/blob/master/builder/frameworks/opencpu-mc60.py#L63

Wiz-IO, 2020-11-25:
I found the bug.... will test and update soon

aakash4895, 2020-11-26:
Nice 

Thank you

Wiz-IO, 2020-11-26:
please test...

only this file is different
https://github.com/Wiz-IO/platform-quectel/blob/master/builder/frameworks/mc60.py

aakash4895, 2020-11-26:
will test and let you know

closed 2020-12-17 by Wiz-IO