Wiz-IO/platform-quectel #42: EC21 OpenLinux Build Error

devwoojin, 2020-12-17:
> Executing task: C:\Users\ssank\.platformio\penv\Scripts\pio.exe run --environment ec21 <

Processing ec21 (platform: quectel; board: ec21; framework: openlinux)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
<<<<<<<<<<<< QUECTEL EC2X MODULE 2018 Georgi Angelov >>>>>>>>>>>>  
CONFIGURATION: https://docs.platformio.org/page/boards/quectel/ec21.html
PLATFORM: Quectel (2.1.3+sha.e3b21dd) > Quectel EC2x module      
HARDWARE: ARM, MDM9607 1200MHz, 10.00MB RAM, 10.00MB Flash       
DEBUG: Current (uart) On-board (uart)
PACKAGES:
 - framework-quectel 2.1.3+sha.cde2255
 - toolchain-gcc-ec2x 1.0.0
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Error: Nothing to build. Please put your source code files to 'C:\Users\ssank\OneDrive\문서\PlatformIO\Projects\EC25 Test\src' folder
================================================================================================== [FAILED] Took 0.85 seconds ==================================================================================================
The terminal process "C:\Users\ssank\.platformio\penv\Scripts\pio.exe 'run', '--environment', 'ec21'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

> Executing task: C:\Users\ssank\.platformio\penv\Scripts\pio.exe run --environment ec21 <

Processing ec21 (platform: quectel; board: ec21; framework: openlinux)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
<<<<<<<<<<<< QUECTEL EC2X MODULE 2018 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/quectel/ec21.html
PLATFORM: Quectel (2.1.3+sha.e3b21dd) > Quectel EC2x module
HARDWARE: ARM, MDM9607 1200MHz, 10.00MB RAM, 10.00MB Flash
DEBUG: Current (uart) On-board (uart)
PACKAGES:
 - framework-quectel 2.1.3+sha.cde2255
 - toolchain-gcc-ec2x 1.0.0
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pio\build\ec21\src\main.o
Archiving .pio\build\ec21\lib_custom.a
'arm-linux-gnueabi-gcc'은(는) 내부 또는 외부 명령, 실행할 수 있는 프로그램, 또는
배치 파일이 아닙니다.
'arm-linux-gnueabi-ar'은(는) 내부 또는 외부 명령, 실행할 수 있는 프로그램, 또는
배치 파일이 아닙니다.
*** [.pio\build\ec21\lib_custom.a] Error 1
*** [.pio\build\ec21\src\main.o] Error 1
================================================================================================== [FAILED] Took 1.03 seconds ==================================================================================================
The terminal process "C:\Users\ssank\.platformio\penv\Scripts\pio.exe 'run', '--environment', 'ec21'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.

running windows 10
how to install arm-linux-gnueabi-gcc?

Wiz-IO, 2020-12-17:
https://github.com/Wiz-IO/platform-quectel/wiki/Framework-OpenLinux#install-notes

devwoojin, 2020-12-17:
it works.!
thank you

closed 2020-12-17 by devwoojin