Wiz-IO/platform-quectel #44: Issue during advanced platform installation

Naveenkato, 2021-02-18:
Hy,
when I paste this command **https://github.com/Wiz-IO/platform-quectel** in advanced platform installation,I continuously getting the error like

`PIO Core Call Error: "Platform Manager: Installing git+https://github.com/Wiz-IO/platform-quectel\r\n\n\nError: Traceback (most recent call last):\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\platformio\\__main__.py\", line 109, in main\r\n cli() # pylint: disable=no-value-for-parameter\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 829, in __call__\r\n return self.main(*args, **kwargs)\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 782, in main\r\n rv = self.invoke(ctx)\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\__init__.py\", line 44, in invoke\r\n return super(PlatformioCLI, self).invoke(ctx)\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1259, in invoke\r\n return _process_result(sub_ctx.command.invoke(sub_ctx))\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 1066, in invoke\r\n return ctx.invoke(self.callback, **ctx.params)\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\click\\core.py\", line 610, in invoke\r\n return callback(*args, **kwargs)\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\platformio\\commands\\platform.py\", line 332, in platform_install\r\n force=force,\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\platform.py\", line 52, in install\r\n spec, silent=silent, force=force, skip_dependencies=True\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_install.py\", line 49, in install\r\n spec, silent=silent, skip_dependencies=skip_dependencies, force=force\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_install.py\", line 97, in _install\r\n pkg = self.install_from_url(spec.url, spec, silent=silent)\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\manager\\_install.py\", line 141, in install_from_url\r\n vcs = VCSClientFactory.new(tmp_dir, url)\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\vcsclient.py\", line 55, in new\r\n src_dir, remote_url, tag, silent\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\vcsclient.py\", line 137, in __init__\r\n self.configure()\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\platformio\\package\\vcsclient.py\", line 146, in configure\r\n result = proc.exec_command([cls.command, \"--exec-path\"])\r\n File \"c:\\users\\katomaran\\.platformio\\penv\\lib\\site-packages\\platformio\\proc.py\", line 117, in exec_command\r\n p = subprocess.Popen(*args, **kwargs)\r\n File \"C:\\Users\\Katomaran\\.platformio\\python3\\lib\\subprocess.py\", line 800, in __init__\r\n restore_signals, start_new_session)\r\n File \"C:\\Users\\Katomaran\\.platformio\\python3\\lib\\subprocess.py\", line 1207, in _execute_child\r\n startupinfo)\r\nFileNotFoundError: [WinError 2] The system cannot find the file specified\r\n\r\n============================================================\r\n\r\nAn unexpected error occurred. Further steps:\r\n\r\n* Verify that you have the latest version of PlatformIO using\r\n `pip install -U platformio` command\r\n\r\n* Try to find answer in FAQ Troubleshooting section\r\n https://docs.platformio.org/page/faq.html\r\n\r\n* Report this problem to the developers\r\n https://github.com/platformio/platformio-core/issues\r\n\r\n============================================================"`

Wiz-IO, 2021-02-18:
is it installed Git?
https://git-scm.com/downloads

Naveenkato, 2021-02-19:
Thanks sir, Problem solved after the installation of Git.
I have another doubt, I wish to work with BG96,,can I use the platformIo ide instead of Quectel SDK??

Wiz-IO, 2021-02-19:
as Text Editor - yes
as independent builder you need to make script ... 
example: https://github.com/Wiz-IO/vscode-quectel-bc66

Wiz-IO, 2021-02-19:
as PlatformIO - read this
https://github.com/Wiz-IO/platform-quectel/wiki/Framework-ThreadX

closed 2021-02-19 by Wiz-IO