Wiz-IO/platform-quectel #46: Debug Output

pcasper90, 2021-03-22:
[image]
[image]
[image]

Hello,
i have problems with the debug output. 
I configured baud rate 115200 and port7 is interface2.

Wiz-IO, 2021-03-22:
seems like a Quectel problem...
reflash module - IF problem again - ask your  local Quectel support

pcasper90, 2021-03-22:
local quectel support does not answer ..

Wiz-IO, 2021-03-22:
https://www.quectel.com/support/contact.htm
https://forums.quectel.com/

pcasper90, 2021-03-22:
I flashed BC66NBR01A04V01 but still the same problem.
Now i created a topic at quectel forum. 

Support Contact formular and support@quectel.com don't answer me ...

Wiz-IO, 2021-03-22:
BC66NBR01A10  recomended

pcasper90, 2021-03-23:
BC66NBR01A04V01 was the only version i found on the internet.
but last night i got contact to quectel support. maybe i can get BC66NBR01A10 from quectel.

nestorayuso, 2021-03-23:
ftp://staging.olimex.com/BC66/

pcasper90, 2021-03-23:
Thank you now i flashed BC66NBR01A10 ;)
But still the same issue for interface 2.

Wiz-IO, 2021-03-23:
what is "interface 2" ?

pcasper90, 2021-03-23:
[image]
[image]

pcasper90, 2021-03-23:
interface 0 for at commands and firmware flashing works fine.
or is it wrong, that i expect the debug outputs on interface2 ?

Wiz-IO, 2021-03-23:
BC66 have only 2 UARTS

pcasper90, 2021-03-23:
ok than i´m not sure how to understand the user guide. how can i see my debug ouput ?

Wiz-IO, 2021-03-23:
#define DEBUG_PORT  UART_PORT0 or 1 and follow the module **UART pins**
USB interface is not supported in OpenCPU

Wiz-IO, 2021-03-23:
work?

pcasper90, 2021-03-23:
not yet .. but just got new infos from quectel. i will let you know when i find my problem ..

Wiz-IO, 2021-03-23:
BC66 OpenCPU UART-s ( in red )
[image]

closed 2021-03-30 by Wiz-IO

pcasper90, 2021-03-30:
I use the bc66-te-b
[image]

Debug Port (Interface 2) is COM10
[image]

But when use putty i can´t read it clear.
[image]

Wiz-IO, 2021-03-30:
[image]

pcasper90, 2021-03-30:
I want to use it without Arduino.

pcasper90, 2021-03-30:
How can i see [APP] begin and [RIL] ready?

Wiz-IO, 2021-03-30:
[image]

pcasper90, 2021-03-30:
ok you use for both the same COM port. now i have a problem while uploading the program
[image]

Wiz-IO, 2021-03-30:
install drivers
[image]