Wiz-IO/platform-quectel #48: How far off is a BG77 framework?

maximevince, 2021-04-01:
Hey @Wiz-IO ,

Thanks for the very nice work. I have been playing a few days with your BC66 framework, which works really nicely.
I was wondering how far off a BG77 port is?

I realize it's a different beast; being based on ThreadX instead of OpenCPU.
Is there anything I can do to help?

Wiz-IO, 2021-04-01:
I have no intention of releasing it - lack of interest....

maximevince, 2021-04-01:
Ok, there is at least some interest, now :)
As I said, I am willing to help if you want.

Wiz-IO, 2021-04-01:
the help is not the problem
BG95/55 is ready before year....
https://www.youtube.com/watch?v=1vF9Lb7n_-k
https://www.youtube.com/watch?v=MdGzIl_6uzA

STB3, 2021-05-27:
Me too, I'm also interested in the BG77. :-)
if you need help or hardware, please tell us.

closed 2021-06-25 by Wiz-IO