Wiz-IO/platform-quectel #49: BG95 compilation error

davctv, 2021-04-21:
Hello @Wiz-IO ,
I'm trying to use your extension for Quectel BG95 (please consider I'm new to platformIO so maybe the problem is just my ignorance).

I have both SDK and compiler (provided by Quectel local FAE).

I created a new project and I have modified c_cpp_properties.json in order to include SDK headers folder, however if I build I get the attached results.

What I do not understand is how I can inform platformIO to use SDK and compiler provided by FAE. Is it related to the fact that BG95 is not "officially" implemented in your platform?

[image]
[image]

Thanks

Wiz-IO, 2021-04-21:
Hi

for now BG95 is not shared yet

davctv, 2021-04-21:
Thanks @Wiz-IO for quick reply.

So, do you have to wait Quectel approval to release it?

If you need SDK and compiler I can send you a private message so you could include them.

Thanks

Wiz-IO, 2021-04-21:
Quectel_BG95_QuecOpen_SDK_Package_V1.1.4 ?

davctv, 2021-04-21:
No, the one the FAE provided me is named "Quectel_BG95_QuecOpen_SDK_Package_V1.1.0"
and compiler LLVM 4.0.3.

Would you say he provided me an old version?
I could ask him if he can provide more recent one.

Thanks

Wiz-IO, 2021-04-21:
ask for new
btw: SDK the version is irrelevant ... important is firmware

davctv, 2021-04-21:
OK I will ask.

Please let me know if you need also FW (what version?) and compiler or anything necessary to include bg95 in your platformIO extension.

Unfortunately Quectel is always missing docs&tools for developers (I use them for about 10 years and it's always a mess)..

closed 2021-06-25 by Wiz-IO

smartechru, 2021-08-04:
Hello @davctv 
I have a BG96 SDK v4.3.2 but was not able to get the EG95 SDK from Quectel.
I would be grateful if you have EG95 SDK.
Thanks in advance.

davctv, 2021-08-04:
Hello @smartechru,

I'm sorry but I don't have SDK for EC95.

Regards,