Wiz-IO/platform-quectel #50: MC60 upload error

luisfear, 2021-07-09:
Hi Wiz!

I love your work.
I'm having a error, in MC60 with openCPU, can you help me?
I tried the versions MC60CAR01A13 and MC60CAR01A12

`BIN SIZE: 0.078125 kB
Configuring upload protocol...
Looking for upload port...
Use manually specified: COM7
Uploading: program
<............................> DONE
Initialising
<..........................................................................................> DONE
Sending MTK Download Agent. Please wait
<....................................................................................................................................> DONE
*** [upload] Explicit exit, status 2

ERROR: APP: MMM`

Wiz-IO, 2021-07-09:
Hi
if this not work use Quectel uploader
I not have module for test

luisfear, 2021-07-12:
With QFlash v4~5 it works without problems. In PlatformIO it works on the serial monitor. But I would like to be able to upload the code from PlatformO but I don't understand the scripts.

```
def openApplication(self, fname, check=True):     
        ASSERT( os.path.isfile(fname) == True, "No such APP file: " + fname ) 
        self.app_name = fname
        with open(fname,'rb') as f:
            app_data = f.read()
        app_size = len( app_data )
        if app_size < 0x40:
            ERROR("APP min size")
        if check == True:
            if app_data[:3] != "MMM": 
                ERROR("APP: MMM")
            if app_data[8:17] != "FILE_INFO": 
                ERROR("APP: FILE_INFO") 
        return app_data  

```

```
Verbose mode can be enabled via `-v, --verbose` option
<<<<<<<<<<<< QUECTEL MC60 MODULE 2018 Georgi Angelov >>>>>>>>>>>>
CONFIGURATION: https://docs.platformio.org/page/boards/quectel/mc60.html
PLATFORM: Quectel (2.1.3+sha.3738509) > Quectel MC60 module
HARDWARE: ARM, MT2503 260MHz, 100KB RAM, 320KB Flash
DEBUG: Current (uart) On-board (uart)
PACKAGES:
 - framework-quectel 2.1.3+sha.cde2255
 - tool-quectel 1.0.0+sha.b65e42d
 - toolchain-gccarmnoneeabi 1.70201.0 (7.2.1)
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
ADD HEADER
Checking size .pio\build\mc60\program.elf
BIN SIZE: 28.703125 kB
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [=         ]   6.4% (used 6552 bytes from 102400 bytes)
Flash: [          ]   0.0% (used 80 bytes from 327680 bytes)
Configuring upload protocol...
Looking for upload port...
Use manually specified: COM7
Uploading: program
<.............> DONE
Initialising
<..........................................................................................> DONE
Sending MTK Download Agent. Please wait
<....................................................................................................................................> DONE
*** [upload] Explicit exit, status 2

ERROR: APP: MMM
```

Wiz-IO, 2021-07-13:
openApplication() test Mediatek header at bin file first bites
if QFlash work with your bin app file - this error is strange....

luisfear, 2021-07-13:
So I understand that the error occurs in the bin verification and not in the upload process itself. Do you have any idea what I can do? What can I try to find out the problem please?

Wiz-IO, 2021-07-13:
open compiled app/bin file and look for MMM & FILE_INFO

luisfear, 2021-07-13:
The .bin file is correct
`MMM@   FILE_INFO    `

If I make this change in MT6261.py, it passes validation (b')

```
if app_data[:3] != b'MMM': 
       ERROR("APP: MMM")
if app_data[8:17] != b'FILE_INFO': 
       ERROR("APP: FILE_INFO") 
```
But I get another error
```
Use manually specified: COM7
Uploading: program
<..........> DONE
Initialising
<..........................................................................................> DONE
Sending MTK Download Agent. Please wait
<....................................................................................................................................> DONE
Uploading application
*** [upload] TypeError : ord() expected string of length 1, but int found
<.......Traceback (most recent call last):
  File "C:\Users\luisfear\.platformio\packages\tool-scons\scons-local-4.1.0\SCons\Action.py", line 1280, in execute
    result = self.execfunction(target=target, source=rsources, env=env)
  File "C:\Users\luisfear\.platformio\platforms\quectel\builder\frameworks\opencpu-mc60.py", line 13, in dev_uploader
    return upload_app(env.BoardConfig().get("build.core"), join(env.get("BUILD_DIR"), "program.bin"), env.get("UPLOAD_PORT"))
  File "C:\Users\luisfear\.platformio\platforms\quectel\builder\frameworks\MT6261.py", line 382, in upload_app
    m.uploadApplication(module, file_name)
  File "C:\Users\luisfear\.platformio\platforms\quectel\builder\frameworks\MT6261.py", line 374, in uploadApplication
    self.da_write_data(app_data)
  File "C:\Users\luisfear\.platformio\platforms\quectel\builder\frameworks\MT6261.py", line 326, in da_write_data
    w = self.crc_word(data[:block])
  File "C:\Users\luisfear\.platformio\platforms\quectel\builder\frameworks\MT6261.py", line 147, in crc_word
    chs += ord(data[i])
TypeError: ord() expected string of length 1, but int found
```

Wiz-IO, 2021-07-13:
python 2 3 ?

luisfear, 2021-07-13:
My platformIO is using its own version, 3.9.4

Wiz-IO, 2021-07-14:
if MMM... FILE_INFO exist... look strange
btw: the script is reverse enginering of mediatek/qflas flash tools

luisfear, 2021-07-16:
But as I said before, it only passes the validation of MMM and FILE_INFO if I make this change 

'MMM' -> b'MMM' 

Is there any possibility that the error occurs in the pyhton script when reading the generated binary?

Wiz-IO, 2021-07-16:
if bin file work with QFlash - application is OK

luisfear, 2021-07-19:
Thank you for your assistance Wiz, you make a very commendable effort, so for the moment I will use QFlash and maybe with some luck, in some platform update in PlatformIO it will be corrected.

Wiz-IO, 2021-07-19:
YW
yep - I need module for test...