Wiz-IO/platform-quectel #59: BG77 SPI with EVB (Evaluation Board)

daviidmart, 2022-11-16:
I’m trying to use an eink display with the SPI protocol but I can’t get it to work I always get the error QAPI_SPIM_ERROR_OPEN_FAILURE. Is this error due to the connections or do I have something wrong configured? I looked in the forum that you have used SPI with the BG95 module that is similar, I hope you can help me.

daviidmart, 2022-11-16:
[image]
[image]
[image]

Wiz-IO, 2022-11-16:
Hi
This platform not support BG77
Please ask support@quectel.com

closed 2022-11-16 by Wiz-IO